logisim-evolution/logisim-evolution #760: .properties files syncronisation

R3dst0ne, 2021-07-02:
It might be a good idea to implement some kind of mechanism that enforces the synchronization of all the properties files.
As you can see in #759 they were quite out of sync.

MarcinOrlowski, 2021-07-13:
We should not auto-synchronize because there may be a case where this would be not really desired. But if you'd make your script i.e. compare "base" `.properties` file with any other and report what is missing (or orphpaned) in the latter and then return non zero RC in such case, then we'd easily make it run as Github Action check

maehne, 2021-07-13:
By default, the English localisation could serve as baseline.

R3dst0ne, 2021-07-13:
> We should not auto-synchronize because there may be a case where this would be not really desired.

I can't think of any situation where we want them to be out of sync. Can you give me an example?

MarcinOrlowski, 2021-07-13:
No, not really, but I see no benefit for automated sync either (we do not automatically fix any issues linters find, even if that'd be doable). The lint can check if these are in sync and if not, flag that. If we include copt&paste steps to sync then that's all what we need.

R3dst0ne, 2021-07-13:
Good point.
As soon as I have some free time I will start working on this.

MarcinOrlowski, 2021-07-13:
Thanks. I think this is two-step effort

- [ ] `*.properties` file sync checker
- [ ] Github Action

For the first one we just need a way to run the script (existing or new one) that takes at least two files (or just two files if that would help), say A and B and then shows how B differs from A (missing keys, orphaned keys). If any diff exists, aside from printing that out, we return non zero RC. Voila. The rest is GHA task (I think you can easily take most of stuff from `.github/workflow/markdown.yml`).

MarcinOrlowski, 2021-07-13:
It'd be great if your linter could also deal with the following:
1. leading spaces (in any case we should not have any in neither A nor B as this is hard to maintain), 
2. * if A ends with punctuation mark (i.e. `.`, `:` or `?`) then if there's B translation, it must follow
3. No leading space is allowed for punctuation marks
4. Unfortunately we need to be able to disable check from 2 (via CLI switch) because of Japanese  (or non-latin in general) translation.

maehne, 2021-07-13:
I would be wary regarding spacing rules in front of punctuation marks. E.g., in French, a non-breaking space is required in front of colons and question marks. Also, French guillemets for quoting are separated by a non-breaking space from the quoted text.

MarcinOrlowski, 2021-07-13:
> a non-breaking space is required in front of colons and question marks

I've seen many french translations and majority did have that. So Is it really "required"? Mind pointing to any reading or so? It'd be really strange (to some degree) to **require** anything like that. It takes more space, so what's the point.

maehne, 2021-07-13:
Look, e.g., [here](https://graphicdesign.stackexchange.com/questions/13697/principles-of-typography-for-different-languages) and [here](https://www.btb.termiumplus.gc.ca/tcdnstyl-chap?lang=eng&lettr=chapsect17&info0=17.07) for common typographic rules.

MarcinOrlowski, 2021-07-13:
Hm, that's interesting. I just browsed the French Wikipedia and indeed they do waste screen space that way. Well, learnt something new :)

MarcinOrlowski, 2021-07-15:
sync tool needs to understand also the case where "in sync" means that B file has the `# ==> key =` for corresponging `key` in A and be told to consider this entry in sync as well as fully translated ones, otherwise we will never have *.properties in sync as we do not have translators on-board 24/7

R3dst0ne, 2021-07-15:
As mentioned in [#761 (comment)](https://github.com/logisim-evolution/logisim-evolution/pull/761#issuecomment-873411408) this is with the current implementation not possible.
The script, in its current form, was never meant to check for synchronism.

I propose rewriting the script with these checks (sync and content) in mind.
(In one week I will have enough free time to do so.)

MarcinOrlowski, 2021-07-16:
I just released `prop-tool` utility (https://pypi.org/project/prop-tool/) which implements all the above, so I think this ticket can now be closed.

R3dst0ne, 2021-07-16:
As long as it is not implemented, this issue should stay open.

PS. It seems the link to the GitHub repository is not working.

MarcinOrlowski, 2021-07-16:
Fixed.

MarcinOrlowski, 2021-07-19:
Just a comment I significantly updated [proptool](https://github.com/MarcinOrlowski/prop-tool) so now it can do more checks on both base files and the translations. It still probably misses some small things to fully fit the task of guarding properties here but that's should happen in days.