logisim-evolution/logisim-evolution #764: Backport README and docs to `master`

MarcinOrlowski, 2021-07-02:
While there's yet no plan for next release, I think we may want to "backport" README and `docs/` from `develop` branch into `master` for "marketing purposes". It looks better, and is more "user friendly". The only blocker for now I see is #745 but once it landed, it should be easy to update main branch. 

@BFH-ktt1 @maehne Opinions?

davidhutchens, 2021-07-03:
I need some time to read through the text and make changes. For example, changing "an" to "a" in a previous PR was an improvement, but still not correct. The words "software" and "hardware" are like "water" in English in that they do not use the article "a". The correct change would be to delete the word. The rest of that sentence somehow trips me when I read it. It should be rewritten, but I haven't taken enough time to figure out how. There are other minor English changes that I would like to make. But I have only skimmed the pages as of now.

So please hold off until I make some edits.

maehne, 2021-07-03:
@davidhutchens: Thanks for your offer to improve the English in our documentation! These improvements should be done first on `develop` to make sure we don't loose them later again inadvertantly.

MarcinOrlowski, 2021-07-03:
@davidhutchens English proofreading, yes please! As this PR is exact copy of files we have in `develop` (with changelog shortened), you can can safely work on `develop/`. I will put this PR on hold and then update it once you done. Or I can close it and you can make both if you wish.

davidhutchens, 2021-07-09:
I made multiple passes over the documents on many days to improve the flow. Then I asked my wife to have a look and she suggested a few more improvements. Thanks, Beth!

I changed the references to the release files to match the actual names.

I added my name to the developers list to encourage others (#590).

It looks like a new 130 character line length restriction is in place. I'll close that PR and fix the files.

BFH-ktt1, 2021-07-12:
Would it be a good moment to make V3.5.1 and merge into master?

MarcinOrlowski, 2021-07-12:
Please don't. I now believe what we really talking here is the next release. Is it major one (4.0) or we still at 3.x  and all the implications like backward compatibility, platform requirements (JDK) etc, etc. I think that what we need to weight is if are at the 4.0 now or still at 3.x. Being 3.x mostly means backward compatibility, but - do we really need to care. Some PRs **could** potentially bite external libraries but we have not tests yet (we **should**).

If I'd be in charge I'd say. Let's make clear we are going for **major**, (potentially) backward incompatible release. If anyone knows any external developer/library/tool that is actively maintained and relies on LSE, let's them know about our intent. Let's do not release before after summer XMAS (we may have nightly builds from `develop` at some point). 

I thinks this is what I think we should make clear in our [ROADMAP](https://github.com/logisim-evolution/logisim-evolution/discussions/695).

MarcinOrlowski, 2021-07-14:
Develop diverged (i.e. JDK14) so that would require additional editing etc. Not worth efforts. Closing.

closed 2021-07-14 by MarcinOrlowski