logisim-evolution/logisim-evolution #771: Bit width drop-down lists

R3dst0ne, 2021-07-04:
> [...] since the increase to 64-bit wide busses, using drop-down lists to choose the bit width and selecting bits is tedious, [...]

By @maehne in #617 [(comment)](https://github.com/logisim-evolution/logisim-evolution/issues/617#issuecomment-859603690)

MarcinOrlowski, 2021-07-05:
I think the good solution for that would be combo widget, containing of widget where user can enter whatever value s/he needs but also accompanying with drop-down "presets" to select for (and I do not think that it makes sense to list i.e. 56 bits. Myh take (based on no random guess): 1-8, 16, 24, 32, 64 (in drop down).

maehne, 2021-07-07:
Yes, that could be a convenient solution.

R3dst0ne, 2022-10-06:
Closing since #1543 got merged.

closed 2022-10-06 by R3dst0ne