logisim-evolution/logisim-evolution #775: References need to be updated

R3dst0ne, 2021-07-07:
I found these outdated references to Logisim-evolution.
- [ ] [sourceforge](https://sourceforge.net/projects/logisimevolution/) download old version
- [ ] [Wikipesia FR](https://fr.wikipedia.org/wiki/Logisim) old version Number, reference to old GitHub URL
- [ ] [Wikipesia DE](https://de.wikipedia.org/wiki/Logisim) old version Number, reference to old GitHub URL
- [ ] [Wikipesia EN](https://en.wikipedia.org/wiki/List_of_free_electronics_circuit_simulators) reference to old GitHub URL

MarcinOrlowski, 2021-07-07:
Corrected Wikipedia entries. As for SF - can **delete** it there?

MarcinOrlowski, 2021-07-07:
Talking about Wikipedia - it might be worth considering to have dedicated page for Logisim-evolution

R3dst0ne, 2021-07-07:
> Corrected Wikipedia entries.

You missed
[image]
in [DE](https://de.wikipedia.org/wiki/Logisim).
Could you update "consulté le" in [FR](https://fr.wikipedia.org/wiki/Logisim) as well.

MarcinOrlowski, 2021-07-07:
Yes I could but I think it's wrong. "Logisim" as such is dead. And it should be clear it is dead so we should not make it look otherwise (and "evolution" does not clearly indicates it's different thingie). I will create Separate entry for "Evolution"

MarcinOrlowski, 2021-07-07:
I am surprised to see there's no English page for Logisim?!

R3dst0ne, 2021-07-07:
I see your point, but leaving a outdated version number is no option eighter.
You could change "Aktuelle Version" to "Nachvolger Versionen". That makes clear that these are successors.

You need to update "consulté le" in [FR](https://fr.wikipedia.org/wiki/Logisim) since https://github.com/logisim-evolution/logisim-evolution didnt exist back in october 2019.

MarcinOrlowski, 2021-07-07:
* I updated DE, but it seems it needs approval to be published now
* Updated FR, should be online now

MarcinOrlowski, 2021-07-07:
We now need to try to have #776 completed. But I not sure I can do this alone anyway

MarcinOrlowski, 2021-07-13:
@reds-heig It seems that SF page is maintained by you:

[image]

Can you please remove it?

CC: @maehne @BFH-ktt1

R3dst0ne, 2021-09-03:
With the 3.6.0 release cumming soon, we really should try to remove the outdated version from [SourceForge](https://sourceforge.net/projects/logisimevolution/).

MarcinOrlowski, 2021-09-04:
Yes, definitely, this one needs to be gone. @TheEvilSkeleton is it something that you can take care of, please?

TheEvilSkeleton, 2021-09-04:
@MarcinOrlowski pardon me for my lack of understanding, but what do I need to take care of?

MarcinOrlowski, 2021-09-04:
Of having logisim-evolution project removed from SourceForge. I thought that you may have some powers there too as it's also under "redsheigvd" type of user as Flakpak was

TheEvilSkeleton, 2021-09-04:
Ah, I'm only involved in packaging in Flatpak and nothing else.

MarcinOrlowski, 2021-09-04:
BTW: if that would be possible, instead of removing we may want to make SF page redirect to github? This seem to be possible and the rationale is that SF page still gets a few downloads a month so why axe working lead.

R3dst0ne, 2021-09-21:
The version number needs to be updated again in: 
- [ ] [Wikipesia FR](https://fr.wikipedia.org/wiki/Logisim)
- [ ] [Wikipesia DE](https://de.wikipedia.org/wiki/Logisim)

maehne, 2021-10-01:
Regarding the outdated Sourceforge page. Taking it down or redirecting it to GitHub can only be done by our colleagues from HEIG-VD in Yverdon-les-Bains. Unfortunately, @reds-heig and his colleagues have been busy more with other things in recent years. @BFH-ktt1 will bring probably this issue up next time he talks to them regarding Logisim-evolution.

maehne, 2021-10-01:
@R3dst0ne: I updated the German and French Wikipedia pages to mention Logisim-evolution 3.6.1.

R3dst0ne, 2022-10-06:
I'm closing, this issue since #1534 features the same (and additional) goals.

closed 2022-10-06 by R3dst0ne