logisim-evolution/logisim-evolution #776: Wikipedia page

MarcinOrlowski, 2021-07-07:
As a follow-up to #775, I created Wikipedia entry of Logisim-evolution as it appears there's no English page neither for LS-E, nor for original Logisim. The page is just quickly made using content from our README, so please take a look and suggest how to improve it. The page is yet to be not published (waiting for a review)

https://en.wikipedia.org/wiki/Draft:Logisim-evolution

MarcinOrlowski, 2021-07-07:
Oh, they rejected my page:

> This submission's references do not show that the subject [qualifies for a Wikipedia article](https://en.wikipedia.org/wiki/Wikipedia:Notability)—that is, they do not show significant coverage (not just passing mentions) about the subject in _published_, _[reliable](https://en.wikipedia.org/wiki/Wikipedia:Identifying_reliable_sources)_, _[secondary](https://en.wikipedia.org/wiki/Wikipedia:No_original_research#Secondary)_ sources that are independent of the subject. Before any resubmission, additional references meeting these criteria should be added (see [technical help](https://en.wikipedia.org/wiki/Help:Referencing_for_beginners) and learn about [mistakes to avoid](https://en.wikipedia.org/wiki/Wikipedia:Common_sourcing_mistakes_(notability)) when addressing this issue). If no additional references exist, the subject is not suitable for Wikipedia.

So it needs more work before re-submission. Assistance required.

MarcinOrlowski, 2021-07-07:
And there's another issue. I copied some texts from README to the wikipedia page but apparently that's not the best idea:

> I have trimmed your draft because the text copied from [1] is licensed under GPLv3 only. GNU-only licenses are not compatible with copying to Wikipedia, per [WP:Compatible license](https://en.wikipedia.org/wiki/Wikipedia:FAQ/Copyright#Can_I_add_something_to_Wikipedia_that_I_got_from_somewhere_else?). Hope that makes sense.

zdimension, 2021-07-28:
I just noticed a Wikipedia page exists for the original Logisim, in French, [here](https://fr.wikipedia.org/wiki/Logisim?oldformat=true)

zdimension, 2021-07-28:
I've updated the [draft](https://en.wikipedia.org/wiki/Draft:Logisim-evolution), can you have a look @MarcinOrlowski @maehne ?

TheEvilSkeleton, 2021-09-04:
Just an FYI, I edited a bit the Wikipedia page. Here's the diff: https://en.wikipedia.org/w/index.php?title=Draft:Logisim-evolution&diff=1042340187&oldid=1041433363&diffmode=source.

TheEvilSkeleton, 2021-09-04:
Just looked around in news sections of search engines, and I found some secondary sources:
1. https://www.google.ca/search?q=logisim-evolution&tbm=nws
2. https://duckduckgo.com/?q=logisim-evolution&t=hx&va=g&iar=news&ia=news

Sadly not so many.