logisim-evolution/logisim-evolution #786: Redrawing of GUI-windows

BFH-ktt1, 2021-07-11:
One of the reasons for slow "simulation" is the fact that for each "change" in logisim the window redraw functions are called. This often leads to a snowball effect on redraw calls, and makes the whole program "slow".
I did not look into all details, but maybe there is a way to "catch" all these redraw requests and only do redraws for example 25 times/s. This should than increase speed and also have a more consistent "simulation feeling".

maehne, 2021-07-11:
Improving the simulation performance by limiting the number of redraws per second sounds promising!

davidhutchens, 2021-07-11:
I had thought of restricting the number of redraws a long while ago. But I didn't investigate it. My thought was to pick a prime number, p, such that with clock speed, c, c/p would be around 20-30. Then draw every p clock ticks. The use of a prime number was to make it unlikely that the circuit would sync with the redraw and never change the visual for changing values. But it might be easier to have another timer trigger the redraws and suppress the request with each clock tick.

It might also be useful to give the user control over the number of redraws per second. Perhaps 5 per second would be good enough for some uses.

Good idea.

maehne, 2021-07-11:
I like @davidhutchens' idea of making the number of redraws per second user-configurable.

crossplatformdev, 2021-11-16:
Hi,

I captured this video with AMD Driver on Windows 10 using Logisim. The clock speed was better than when I remade the project on Logisim Evolution (the first schematic ran above 2Khz, with Logisim Evolution I can barely achieve 120Hz, with the same set up).

https://www.youtube.com/watch?v=UP6tO8x5I5A

The schematic is authored by me and it is Open Source, just in case Logisim devs need it to help me achieve the max hertz speed on Logisim pro. 

https://github.com/ElijaxApps/Libre-8

Let's solve this issue please.

maehne, 2021-11-22:
Thanks @crossplatformdev for your test case! I think we all agree that any improvement w.r.t. the simulation speed is very welcome, as are PRs in this area. Our bandwidth during the teaching term is unfortunately limited and we have to focus first on fixing bugs, which hamper our students' work.

rnayabed, 2025-01-21:
Perhaps we could have a "high performance" mode where the data lines and most of the components are not re-drawn?

crossplatformdev, 2025-01-25:
Search this repository for every Thread.sleep(milis) and delete them: You don't need them.

With a custom logisim with all the Thread.sleep(milis) calls commented, I can reach speeds of 1.4 Khz.

[image]

Thread.sleep(milis), as it is intended, sleeps a minimum of 1 MILISECOND. 
Currently, when running at 512 hertz, this function is called 512 times, sleeping a total of 512 miliseconds, then not obtaining EVER more speed that ~500hz.

With the calls commented, the thread does not sleep, calling tick again inmediately afterwards. Then in the file Clock.java in 
logisim > std > wiring, try this faster tick, as a demonstration (breaks the phasing and the ammount of cycles per fase, but is undoubtable faster, as it gets rid of innecesary mod operations, which are very slow):

```
    boolean updateTick(int ticks, AttributeSet attrs) {
      /* All this variables should be class fields, to avoid
      /* Recalculating it on each tick (!!!)
      int durationHigh = attrs.getValue(ATTR_HIGH);
      int durationLow = attrs.getValue(ATTR_LOW);
      int cycle = durationHigh + durationLow;
      int phase = ((attrs.getValue(ATTR_PHASE) % cycle) + cycle) % cycle;
      boolean isLow = ((ticks + phase) % cycle) < durationLow;
      Value desired = (isLow ? Value.FALSE : Value.TRUE);
      if (sending.equals(desired)) return false;
      sending = desired;
      */
      sending = (sending.equals(Value.FALSE)) ? Value.TRUE : Value.FALSE;
      return sending.equals(Value.FALSE);
    }
```

crossplatformdev, 2025-01-25:
These two pictures, from a video of 10 seconds, were shot with base logisim-evolution 3.9.0:

- 1.9 Khz running a clock alone.
[image]
- 734Hz running Libre 8 schematic.
[image]

After the two changes above mentioned, same simulations with my custom logisim-evolution 3.9.0:
- 2.25 Khz running a clock alone.
[image]
- 885Hz running Libre 8 schematic.
[image]

crossplatformdev, 2025-01-25:
Forked and created `nothreadsleep` branch. 

Removed offending Thread.sleep calls. Consistent max speed 1.92Khz.

[image]

davidhutchens, 2025-02-17:
I realized in working on #2129 that it might be fairly easy to limit the frame rate of the drawing while running the simulation. I'll see if that gives any meaningful speed increases.

maehne, 2025-02-18:
Reopen issue, as I think that it was closed accidentally.

@davidhutchens: I am looking forward to your investigation what speed increases limiting the frame rate will allow!

crossplatformdev, 2025-03-02:
Hey, I thought a little about it, and now I consider it would be fair for everyone if I offer a reward* to the developer, or the project of its choice**, to whoever makes Logisim-Evolution to reach Mhz, or preferably, if it is possible, Ghz range.

I really need this feature to make a better Libre 8, if some developer can dedicate its efforts to this, I am more than willing to give you the beans. :-)

Libre is a 8 bit computer schematic that runs in Logisim, see it in action here:
[https://youtu.be/l9oFCmCrPmg
](https://www.youtube.com/playlist?list=PLlva0633MgdbbNI8wojc-jN8GsXoJPEUM)

Currently it can do a lot of things! Conditional Flow, Functions, with Arguments pass and Recovery on dest., Integer math, I/O instructions, and even a GPU that can reproduce video and render 2D games!

Sources:
https://github.com/crossplatformdev/Libre-8

*Paypal, Kofi, Patreon, Bizum or SEPA.
**Public addresses only.

maehne, 2025-04-26:
@BFH-ktt1 and @davidhutchens: Can this issue be closed due to the recent merges of PRs #2170, #2185, #2186, #2192?

davidhutchens, 2025-04-26:
@maehne: There is one thing that has not been addressed to date. We do not give the user the ability to set the frame rate for redraws. There might be a very modest increase in speed by having a much lower frame rate. But just adding the option to control that rate is not good since the rate currently applies to all redraw requests, including those from user actions. So dragging a component looks awful with a lower frame rate. To allow changes to the frame rate would therefore require separating the simulation generated redraw requests from the user generated requests. It is not clear it is worth the effort. But that is why I have not closed this yet.

maehne, 2025-04-26:
Thanks for the quick response @davidhutchens! OK, then we'll let you decide when this issue should be closed.

crossplatformdev, 2025-05-25:
@davidhutchens thanks for all your efforts, first of all. 😀

Libre 8 has passed of running 440Hz to running at 10kHz now, great job! I am wondering if there's still something to be merged to finally reach those 2048 kHz that are displayed on the frequencies menu (as I say, limit seems capped to 10Khz, when running Libre 8 + Megawatts video; and I tried other videos, with same result).

I have created #2240 to see exactly why Libre 8 does not reach the full speed, and what can we do with my schematic to use it as a benchmark against Logisim, and make the simulator run as fast as we can 😀

And certainly, you earned the reward I proposed, already by now. I won't insist anymore, but FYI I am more than happy and still willing to invite you to some beans of coffee for the hard work, if you agree.

At least accept a big thank you, again!

crossplatformdev, 2025-07-22:
Hi everyone,

Hey @davidhutchens, I would like to ask about the merges that affected speed performance.

You mentioned that in simple schematics you could achieve speeds of 2 MHz. I’ve been able to achieve this randomly by leaving the #libre8 schematic simulation running for a long time with a program that jumps back to the beginning. I suspect that the second time my computer ran the program, the simulation was cached and the results were retrieved faster. However, this only happens when using the TTY output, not the RGB video output.

I know that internally, LE must handle many events that are sometimes chained, need sorting, and so on. But honestly, since my computer processes only one byte at a time, not all components in my schematic are always fully utilized.

In any case, I’m running Windows 10 with the following Java version:
```
> java -version  
java version "21.0.7" 2025-04-15 LTS  
Java(TM) SE Runtime Environment (build 21.0.7+8-LTS-245)  
Java HotSpot(TM) 64-Bit Server VM (build 21.0.7+8-LTS-245, mixed mode, sharing)
```
My question is: when I noticed a 20× speedup, the maximum frequency I reached with cached results was 1.31 MHz, whereas on the first run it struggled to reach around 10 kHz.

1. Would it be possible to warm up a certain amount of cycles before a simulation? Could this option be added to the toolbar menu and be uncapped via a prompt?  
2. Does this ring a bell regarding what could be preventing the RGB component from exceeding 10 kHz?

Thank you for your efforts.

Elías A.

maehne, 2025-07-23:
@crossplatformdev: IMHO, tuning performance for niche use cases like modelling entire computer systems at the gate level is out of scope for Logisim-evolution as a teaching tool. Our resources are limited. Therefore, we should focus on issues affecting more users, e.g., improving the documentation and modernising the help system.

Much higher gains in terms of simulation performance are achievable by modelling at transaction level instead of gate level, e.g., using our SoC components.

commit 9dfbd1fd (2025-02-18): Close #786