logisim-evolution/logisim-evolution #789: Bumping min JDK to JDK 14

MarcinOrlowski, 2021-07-11:
I'd like to propose bumping min. version of JDK from current 11 to at least 14, mainly for improved [switch()](https://openjdk.java.net/jeps/354) and [Text Blocks](http://openjdk.java.net/jeps/355). Any reasons for not doing that?

@BFH-ktt1 @maehne @davidhutchens

maehne, 2021-07-11:
From my side, I am fine with bumping the minimum JDK version to 14 or above. Main reason for keeping support for OpenJDK 11 would be that it is still the latest LTS version, which is widely distributed in Linux distributions. (E.g., until the beginning of this year, it looked like the upcoming Debian Bullseye would only ship OpenJDK 11. Now, it seems that it will also ship OpenJDK 17.) However, since the advent of `jpackage` in OpenJDK 14, we can finally build self-contained installer packages again, which don't require the installation of a JDK as a dependency. Therefore, this is less important now. IMHO, we should not strive to support older OpenJDK version at all cost. If there are features in newer JDKs, which we deem of value to be used in Logisim-evolution, we should allow ourselves to move to this version. We should only keep support for older JDK versions, which comes basically for free, as it has been till now with OpenJDK 11.

Asking all main contributors for their opinion before such a move is polite and ensures broad acceptance of such a move and we probably should do it in the same way when we want to bump JDK versions in the future.

davidhutchens, 2021-07-11:
I have no problem bumping the minimum JDK version to 14. Another reason to do so is that there were some bugs on the Mac that were fixed in JDK 13; so Mac users benefit from using v13 or higher. Also, we need JDK 14 to build the packages. It would be cleaner to have JDK 14 everywhere.

The main reason not to, as @maehne noted: OpenJDK 11 is the latest LTS version. My search shows OpenJDK 17 will be an LTS version when it is released Sept. 14. I decided to have a look at OpenJDK 17. I downloaded the current build, cleaned and built Logisim-evolution with it, and ran some examples. There were no problems. I did not need to upgrade any components of the build system.

One possible issue with OpenJDK 17 that we will need to address is that there are now two versions for the Mac, x64 and AArch64. My x64 version builds the same dmg with the same name as before. But the AArch64 version should build a version for the Mac M1 processor. I am not sure what the dmg will be named on that version. We will likely wish to build both and include both in our releases and update the documents accordingly. (Sorry to hijack your topic, @MarcinOrlowski).

I favor moving the minimum to JDK 14. The only question in my mind is when. Now, or after OpenJDK 17 is released. I don't see a strong reason to wait. But I don't know how many people are building from source on systems that only (or by default) use LTS versions.

MarcinOrlowski, 2021-07-11:
> Sorry to hijack your topic

There's nothing to be sorry about! This ticket is just for this kind of discussions/thoughts/feedback/notes (also this is great example why we should have [tickets prior each PR](https://github.com/logisim-evolution/logisim-evolution/issues/753#issue-934725805)). It would just be good to use more recent JDK version for various benefits (incl. JVM improvements, not only the language features) so I am glad to see "it's in general good idea but..." attitude :) Luckily, we absolutely do not have to do it right now, which gives us the opportunity to plan the transition carefully with all pro and cons weighted. So let's keep going.

davidhutchens, 2021-07-12:
All I meant by the hijack comment is that I realized that paragraph (about Mac versions) is a bit off-topic. All is good.

maehne, 2021-07-12:
If bumping OpenJDK to version 14 fixes bugs on macOS, I am in favour to do this change without much further waiting. It reduces the risk that issues related to these bugs get created. What do you think @BFH-ktt1?

BFH-ktt1, 2021-07-12:
For me this is okay. We should just document it.

davidhutchens, 2021-07-12:
The macOS bug I was referring to is #219. It works correctly when run on Java 13 or higher, no matter what we have as our minimum version. So pushing the minimum doesn't really help most users, who are already using something higher. I don't  actually know if that bug fix was pushed back into JRE/JDK 11 LTS (the bug page itself does not say so); and it is unlikely to have been pushed into JRE/JDK 12.

Anyway, I have no objection to moving forward with the change to JDK 14+ whenever you all are ready.

MarcinOrlowski, 2021-07-12:
I created PR that switches to JDK14. Aside from this and docs update I also changed our action builder to use just one JDK14 instead of lowest and highest supported as we had so far (11, 16). There's relatively no risk nor need for that and it saves us some time.

PR #797 (2021-07-13): Increases min JDK version to 14. Closes #789
Increases min JDK version to 14. 

Closes #789
closed 2021-07-14 by MarcinOrlowski