logisim-evolution/logisim-evolution #802: Do not save unused libraries in the circuit file

maehne, 2021-07-13:
There's no logic to skip libraries that are unused, but were active when the Logisim circuit file was created. This will cause a File Error that the library is not available (cf. to issue #218). Logisim should not save references to unused libraries in the circuit file.

BFH-ktt1, 2022-08-26:
The option is now there, can this issue be closed?

R3dst0ne, 2023-02-05:
@BFH-ktt1 is there a reason why you reopened this issue?
As you mentioned, this functionality is now present. (has to be activated manually)

maehne, 2026-09-28:
Sorry @hewzhew for the delayed feedback on this issue, I would opt for choice two, i.e., the minimum visible should be implemented first, as discussed in issue #1200.