logisim-evolution/logisim-evolution #810: Renaming master branch to main

R3dst0ne, 2021-07-13:
Information why we should consider it can be found [here](https://github.com/github/renaming/).

MarcinOrlowski, 2021-07-13:
Please be serious.

maehne, 2021-07-14:
I am not in favour to do this urgently. People are used to the default name to be named `master`. Many words have different meanings depending on the context. In this case it is the same, see, e.g., [Merriam-Webster](https://www.merriam-webster.com/dictionary/master). In our case, most fitting meaning is 5b "an original from which copies can be made". In our case, that branch serves as master for building our binary releases for Logisim-evolution. Furthermore, people who have already cloned our repository would be affected by this change by encountering unexpected Git error messages, which would probably cause unnecessary reported issues, we have to respond to.

If this renaming should gain wider practice and the migration burden gets further smoothed-out through improvements in `git`, we may consider it at a later point if our users show sufficient support for this change.

R3dst0ne, 2023-04-15:
I reopened this issue because IMO, the concerns @maehne had are not that relevant anymore.

> People are used to the default name to be named master.

Since October 1, 2020 GitHub uses `main` as default branch name.
And git introduced `--initial-branch=<branch-name>`(v2.28) and the `init.defaultBranch`(v2.30) configuration variable to specify the name for the initial branch in the newly created repository.
`main` is quite commonly found now.

> Furthermore, people who have already cloned our repository would be affected by this change by encountering unexpected Git error messages, which would probably cause unnecessary reported issues, we have to respond to.

If #1552 happens, we probably are going to face similar problems.

> If this renaming should gain wider practice and the migration burden gets further smoothed-out

GitHub did a lot to make this process smother and provided good documentation on what to do. See https://github.com/github/renaming/ and the [GitHub Docs](https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/managing-branches-in-your-repository).

closed 2023-07-20 by R3dst0ne