logisim-evolution/logisim-evolution #815: java v16: themes don't work

lightonflux, 2021-07-14:
When using java v16 instead of e.g. v11 then themes like solarized light, dracula and intellij don't work, and lead to the application not starting.

## Java version:
```
./java  --version 
openjdk 16 2021-03-16
OpenJDK Runtime Environment 21.3 (build 16+36)
OpenJDK 64-Bit Server VM 21.3 (build 16+36, mixed mode, sharing)
```

```
./java  -jar ~/Downloads/logisim-evolution-3.5.0-all.jar
Exception in thread "main" java.lang.IllegalAccessError: class com.github.weisj.darklaf.ui.optionpane.DarkOptionPaneUI (in unnamed module @0x66ff6b1) cannot access class sun.swing.DefaultLookup (in module java.desktop) because module java.desktop does not export sun.swing to unnamed module @0x66ff6b1
	at com.github.weisj.darklaf.ui.optionpane.DarkOptionPaneUI.createButtonArea(DarkOptionPaneUI.java:42)
	at java.desktop/javax.swing.plaf.basic.BasicOptionPaneUI.installComponents(BasicOptionPaneUI.java:205)
	at java.desktop/javax.swing.plaf.basic.BasicOptionPaneUI.installUI(BasicOptionPaneUI.java:159)
	at java.desktop/javax.swing.JComponent.setUI(JComponent.java:685)
	at java.desktop/javax.swing.JOptionPane.setUI(JOptionPane.java:1851)
	at java.desktop/javax.swing.JOptionPane.updateUI(JOptionPane.java:1873)
	at java.desktop/javax.swing.JOptionPane.<init>(JOptionPane.java:1838)
	at java.desktop/javax.swing.ProgressMonitor$ProgressOptionPane.<init>(ProgressMonitor.java:161)
	at java.desktop/javax.swing.ProgressMonitor.setProgress(ProgressMonitor.java:291)
	at com.cburch.logisim.gui.start.Startup.autoUpdate(Startup.java:593)
	at com.cburch.logisim.Main.main(Main.java:68)

```

MarcinOrlowski, 2021-07-14:
Looks like dupe of #566 which is already addressed in `develop`. Can you please check?

maehne, 2021-07-14:
I confirm that this bug is already addressed in the `develop` branch by PR #619, which replaced the DarkLaf look and feel with FlatLaf. @lightonflux: For the moment, you will have to build yourself Logisim-evolution from `develop` to use one of the new look and feels until we'll release a new version, which will very likely happen the latest in the early autumn when our new academic term starts. Currently, `develop` is evolving fast due to considerable refactoring efforts, which will require some more thorough testing before making a new official release.

closed 2021-07-14 by maehne