logisim-evolution/logisim-evolution #818: Nightly builds

MarcinOrlowski, 2021-07-14:
I created (experimental kind of) nightly builder that is expected to be run daily (cron) and create new builds if there's any activity on develop branch during last 24 hours. For now it builds *.deb, shadow jar and source jar. Once it is stable we can add more packages.

maehne, 2021-07-14:
Great! This feature has been for long on our internal wish list! Building RPM packages should be straightforward since it only requires installation of `rpm` on Ubuntu. MSI and DMG packages need to be built on Windows 10 respectively macOS though. Nevertheless, already providing a Shadow JAR would facilitate for our users a lot testing bug fixes.

MarcinOrlowski, 2021-07-14:
Updated with RPM support. As for macOS I may need some guidance: what [runs-on](https://docs.github.com/en/actions/reference/workflow-syntax-for-github-actions#jobsjob_idruns-on) should be used for the worker? and which gradle task: `createApp` or `createDmg`?

davidhutchens, 2021-07-14:
It should use createDmg. It looks like macos-latest would work, or macos-10.15 (right now it says they are the same).

MarcinOrlowski, 2021-07-14:
Is Java installed out of the box? Do I need to install any extra devtools? if so, can you provide copy-paste lines that set up all the dev env via shell so `./gradlew createDmg` can work?

davidhutchens, 2021-07-14:
According to: https://github.com/actions/virtual-environments/blob/main/images/macos/macos-10.15-Readme.md OpenJDK 14.0.2 is installed as well as the Xcode development tools.

The Java environment variable is: JAVA_HOME_14_X64. Use that to set JAVA_HOME.

It sounds like that should be all you need to build the .dmg package.

MarcinOrlowski, 2021-07-14:
Seems it worked. BTW: is there any reason why you did [this](https://github.com/logisim-evolution/logisim-evolution/blob/bb6fd01e5218583ba0e33e5ef0745e02286e74de/build.gradle.kts#L84):

    val uppercaseProjectName = projectName.substring(0,1).toUpperCase() + projectName.substring(1)

instead of having all lowercased?

MarcinOrlowski, 2021-07-14:
@davidhutchens @maehne here's DMG for you to test: https://github.com/logisim-evolution/logisim-evolution/actions/runs/1031623910

MarcinOrlowski, 2021-07-14:
Except for Windows all platforms are now supported on nightly. Here is output from example run:

https://github.com/logisim-evolution/logisim-evolution/actions/runs/1031646246

davidhutchens, 2021-07-14:
The name that jpackage uses for the appname gets incorporated into the application in various ways including the Info.plist file. Using a lower case letter looked strange mixed with the other references within the program. That change makes the app look better. jpackage also uses the appname to name the .app directory and the .dmg file.

The names that you changed in build.gradle.kts do not SET the name. The name is determined by jpackage from the appname. Those names are used to determine if it is necessary to rebuild it in gradle. So your change likely breaks the test for rebuilding.

When jpackage was first introduced into our build process, it built the .dmg directly without saving the .app. I broke apart the steps to fix some issues (after it creates the .app in createApp). It might be possible to name the dmg without renaming the app. I haven't actually looked at it.

The dmg you asked me to test works! Is it necessary to zip the dmg?

MarcinOrlowski, 2021-07-14:
> So your change likely breaks the test for rebuilding

Hm, ok. I will take a closer look later then. Thanks. 

As for ZIP archived artifacts. It is Github's:

> However, when a workflow run finishes and an artifact is downloaded from either the UI or through the download api, a zip is dynamically created with all the file contents that were uploaded. **There is currently no way to download artifacts after a workflow run finishes in a format other than a zip or to download artifact contents individually**

https://github.com/actions/upload-artifact#zipped-artifact-downloads

davidhutchens, 2021-07-14:
Actually, with respect to package names, jpackage does weird things including using _ or - on different linux packages. I wonder if we might add a final step to each build that moves the package into a name format that we make consistent across platforms.

MarcinOrlowski, 2021-07-14:
And here comes Windows support:
https://github.com/logisim-evolution/logisim-evolution/actions/runs/1031904141

MarcinOrlowski, 2021-07-15:
> I wonder if we might add a final step to each build that moves the package into a name format that we make consistent across platforms.

That's actually a good idea and I think bests place to normalize the names. Could you please take care of this?

maehne, 2021-07-15:
**Attention:** As far as I remember at least DEB and probably RPM files have their own naming convention, which should be followed. These impose where dashes and where underscores separate file name parts and how the package version should be indicated. If these are not respected, there may be errors generated by the package managers.

MarcinOrlowski, 2021-07-15:
Closing this one as nightly is merged. I created #821 to follow-up with name normalization.

MarcinOrlowski, 2021-08-13:
Reopening as I do not see nightly worker doing anything. Can I be given rights to project Actions to investigate/debug or anyone (@BFH-ktt1?) can take a look there?

BFH-ktt1, 2021-08-13:
@MarcinOrlowski , I gave you maintainer rights. We will see with the roadmap discussion how to handle the rights. Let me know if this works for the debugging

PR #877 (2021-08-14): Reworked nightly builder
Fixed a bug in git history handing that could cause worker script to fail silently under some circumstances. Also prevented future silent failures and did some cleanup.

I also restored `pull_request` trigger because `schedule` only works if worker is part of main branch (which is not the case unless next release). This is not really a big deal really. We will have occasionally more than one "daily" build per day (with file named the same but different content though) but this is the only minor issue. I could add hash to filename but I do not think it is really worth t [... cut]
closed 2021-08-14 by MarcinOrlowski