logisim-evolution/logisim-evolution #821: Normalize package file names

MarcinOrlowski, 2021-07-15:
We need to normalize how package files (*.deb, *.rpm, *.dmg, *.msi, *jar) are named as there are might be platform specific requirements that require specific naming while building. Yet, final **file** can most likely be safely renamed to our liking and this ticket is all about that.

@davidhutchens [proposed](https://github.com/logisim-evolution/logisim-evolution/issues/818#issuecomment-880228963) tweaking gradle tasks to deal with that which should be perfectly sufficient.

I believe that we should stick to the following rules

1. while name is lower-cased
2. only [a-z][0-9], `.`, `-` and `_` are allowed
3. we should have just singe `.` in the name (unless there's a valid reason to have more)

I propose segments to go like this:

    <NAME>_<VERSION>[-<PLATFORM SPECIFIC>][_<ARCH>].<SUFFIX>

1. `NAME` nad `VERSION` should be same as in `gradle.properties` (`NAME` lower-cased).
2. `<PLATFORM SPECIFIC>` is usually distribution specific version suffix (i.e. `-1` for *.deb etc). This is questionable though as file name can be really anything and we can safely drop that suffix for no penalty. We may need to re-instantiate it if we dedicate to run our repository (like PPA) but that's not happening now anyway.
3. `<ARCH>` is platform specific and mostly matters for Linux (and maybe macOS. Not sure about Windows though). It should be `amd64` for Debian but `x86_64` for RPM. Not sure about macOS again.
4. `<SUFFIX>` should be regular `deb`, `rpm`, `dmg`, `msi`, `jar`. 

That should give us names like:

```bash
logisim-evolution_4.0.0-1_amd64.deb
logisim-evolution_4.0.0_all.jar
logisim-evolution_4.0.0.msi
```

We also have nighty builds and the names of produced artifacts will be different a bit. That's irrelevant for the context of this ticket as names are set by the nightly builder, but I leave a note here. Currently nighty are named:

    <NAME>_<BRANCH>_<YYYYMMDD>[_<ARCH>].<SUFFIX>

i.e.

    logisim-evolution_develop_20210714_amd64.deb

davidhutchens, 2021-07-15:
Given @maehne's comment in #818 about the package name assumptions in Linux distributions, this sounds like a bad idea with respect to the main packages. So I withdraw my suggestion, and propose we leave the package names as they are.

We could create special nightly build names, if that sounds appropriate to everyone. I'll wait for more comments.

MarcinOrlowski, 2021-07-15:
As definitely there are some guidelines, I am pretty certain that what is proposed above pretty much follows common practices. While I cannot tell for all the cases, I am pretty sure i.e. *deb is compliant. Other are also easily verifiable. The only thing are the nightlies, but that's definitely not a problem either. We do not need to follow strictly unless we want to contribute to main repository. I am also bit skeptical that package manager cares package file names.

MarcinOrlowski, 2021-07-16:
Debian naming convention: https://www.debian.org/doc/manuals/debian-reference/ch02.en.html#_debian_package_file_names which means, for the binary package (*.deb): `<package-name>_<upstream-version>-<debian-version>_<architecture>.deb`. So we are good here.

RPM naming convention: https://docs.oracle.com/en/database/oracle/oracle-database/19/ladbi/rpm-packages-naming-convention.html so i.e. `The naming convention for RPM packages is name-version-release.architecture.rpm`. We are bit off here as the do not use underscore. Easy to fix on our side.

davidhutchens, 2021-07-16:
Our rpm package has the underscore in the middle of the architecture description: x86_64, so our current name conforms.

Note that all our names are the ones produced by Java's jpackage utility. We are not changing them. So it isn't surprising that they match the expectations.

I think we should just leave the names alone since they are what the platforms expect.

MarcinOrlowski, 2021-07-16:
I good with that. My only pain was that for macOS we had capitalized first letter of filename, which made exception for nightly. And exceptions are bad.

davidhutchens, 2021-07-16:
I have a version of build.gradle.kts that builds the Mac application with an uppercase letter (so it looks right when running), but uses a lowercase letter on the dmg name. That is slightly odd, but not too bad.

MarcinOrlowski, 2021-07-16:
Maybe I do not understand the problem with this. Please educate: what I want is whatever final package **fille** we buiild its name is all lowercased. That's simplifies the further steps. What is the probilem with macOS is DMG **file** name is all lowercased?

davidhutchens, 2021-07-16:
I said it is slightly odd. It is not a particular problem. My previous comment was intended to say that I have that working and I can do that when I fix the #827 issues.

davidhutchens, 2021-08-13:
I think this issue can now be closed.

PR #829 (2021-07-28): Fix gradle build, change mac .dmg name to lower case, and update README
This PR does the following:

Fixes the gradle build problem noted in #827.

Changes the macOS .dmg package name to lowercase, but leaves the .app name as uppercase so the user interface is not changed. Closes #821.

Updates the README to change the case of the dmg.

Updates the README to add a note for macOS users regarding the initial permission to run and the possible firewall question about internet access as discussed in #747.

Improves other English in the README.

I also added a commit to indent the build.gradle.kts  [... cut]
closed 2021-08-13 by BFH-ktt1