logisim-evolution/logisim-evolution #824: Separate documentation from code

MarcinOrlowski, 2021-07-15:
We shall consider separating documentation from code. This brings several pros and cons:

Pros:
* we can update docs any time, independently from app release cycles
* instant fixes
* community can easily address typos, grammar and other issues

Cons:
* requires on-line access (we can still have some form generated form online versions (i.e. PDF from HTML))
* maintaining multilingual docs is challenging

@BFH-ktt1 @maehne @davidhutchens

jxu, 2021-07-19:
Docs on which components? I am a fan of maintaining documentation for internals (like specific classes) with the code itself

MarcinOrlowski, 2021-07-19:
"from code" means from app. That would allow independent release cycles, faster updates, easier contributions. This ticket is related to end user docs, not code docs.

mbaillif, 2021-07-22:
Some points to defend or not the idea.

a) Currently the end user documentation is in HML and we use javahelp2 from Sun which only supports HTML 3.2 and CSS1 (1987).
b) Lately I have modified this html so that it is still compatible with javahelp, but also with the more modern browsers which could make it a bit strange.
These changes have improved the cosmetics of the pages, but it makes it more difficult for anyone to modify the text.
C) most of this documentation is a rehash of what already existed in logisim.

I think it would be nice to make this task easier. (Markdown/Ascidoc...) and an application for online editing or not.

However, from what I observe, I don't think there is much spontaneous documentation. And someone will have to hunt for information.
The laziness starts already in the development. If each component was accompanied by a small text(ascii) presentationi it would make it easier for documentation.
For example like in gui.hex.hexfile.java

Open questions.

- What help is there in Logsim-evolution offline?
- is there a more modern alternative to javahelp2?
- Depending on the case: hosting, domain name and costs of that?
- The transfer of the current documentation to the new structures.
- Moderation, control, volunteers to follow the different languages.

jxu, 2021-07-22:
Some answers

> What help is there in Logsim-evolution offline?

The user docs are available with the program? Is that what you mean by offline?

> Depending on the case: hosting, domain name and costs of that?

GitHub pages offers free hosting for static pages which should be sufficient.

MarcinOrlowski, 2021-09-06:
> I had some discussion with @BFH-ktt1 and he said we should strive for continuing to deliver the user and library documentation as part of the program package so that it remains accessible when using Logisim-evolution offline, e.g., in exam settings.

I do not mind offline docs but here's my take: I'd love to have **one** source documentation from which we can **generate** anything needed, incl. in-app javahelp pages too. Even if that would not be possible too easy, we could still be able to generate i.e. offline PDF or even ZIP with HTMLs (or single HTML page (massive, but searchable easily) + ZIPed HTML pages). PDF readers + modern browsers are far superior to ancient javahelp so javahelp should not even be a part of the equasion (IMHO).

BFH-ktt1, 2021-09-06:
My take on the issue:
1) I think that it makes sense to make a separate git repository for the documentation and to use for example Markdown for the text. This would simplify the task of "spontaneous" documentation as @mbaillif called it. For the moment, each time I go into the doc directory I get a nightmare and have to figure each time out where, what, how.
2) From the doc in point 1) we should than have workers that generate the web-contents and pdf.
3) We should also have a worker that imports the actual doc inside the logisim-jar for "offline" usage. This does not have to be java-help, I can imaging that there are java projects that can browse and display pdf documents, or maybe opens the pdf-document direct in a browser, or shows a html-contents.

MarcinOrlowski, 2021-09-06:
> or maybe opens the pdf-document direct in a browser, or shows a html-contents.

Yep, that should suffice. We will loose contextual help but the benefits are far bigger than this loss.

So before we move on, let's see how we want the separation to look like and what is "documentation language/system" of choice. Will markdown suffice? If so, this can be converted to PDF and HTML with ease.

MarcinOrlowski, 2021-09-06:
We also need to think about multiple language versions. Do we want them in sync (i.e. filenames must be the same etc)

MarcinOrlowski, 2021-09-06:
As for Latex - it's complicated for what we have. Markdown is sufficient. You got basic structure, basic elements, links and images. Do we need more?

jxu, 2021-09-06:
LaTeX in my experience is good for typesetting whole manuals or math/engineering papers. For code documentation which probably won't have much math, Markdown is sufficient and lighter weight.

BFH-ktt1, 2021-09-06:
I think the format we choose should be the "simplest" that suite "our purposes", so if it is Markdown it's fine by me.
As for the multiple language version, I think that this is one of the "strong" points of logisim. It is just that we have to find "native speakers" that are willing to join this effort. For English, French, German, Polish, and Dutch (maybe Portuguese) we already have them, what about the others (Russian, Japanese). I think that the property languages should match also the documentation languages.

BFH-ktt1, 2021-09-06:
@jxu: We are not talking about the documentation of the source code, but about the "user guide" (documentation), like "how to become a logisim user". I got the feeling that you think we talk about the documentation that tells how the source-code is build up, which is in fact something that we also have to discuss, as it is not "really" present.

jxu, 2021-09-06:
I am more of a fan of online HTML docs (like that of plotly or tidyverse or MDN) than a PDF guide. For that case Markdown should suffice. I guess if we really wanted to go in-depth with the program, there could be a whole MediaWiki on it :)

MarcinOrlowski, 2021-09-06:
I renamed thw thread to make it more clear. Source code docs or architecture docs - this is different story

MarcinOrlowski, 2021-09-06:
From other sie I noticed some element in docs are separately emphasized using html/css. We will loose that with markdown. Is it a problem? Or maybe there's extended markdown we clld use?

jxu, 2021-09-06:
Nah, there are tools that convert Markdown to HTML like pandoc. Then you can style CSS from there.

MarcinOrlowski, 2021-09-06:
> Then you can style CSS from there

But how? If manually - forget it. We must have just one file we edit. No manual postprocessing

I just realized that you are allowed to put HTML markup in markdown file so you can still have `<em>` or `<span>` if needed I think.

jxu, 2021-09-06:
Not sure what you mean by manually. You can use a premade reasonable CSS or tweak your own

MarcinOrlowski, 2021-09-06:
I mean that any information (markup-ish) must be in the source file. CSS (as part of presentation layer) can be edited separately (even manually) but you must be able to generate final version of certain type of document directly from sources co content must be self-contained. Any further changes or post-processing are fine if needed, but that must be fully automated process. But it'd be good to be able to add i.e. own markdown tags, indicating extra highlighting. We'd then

jxu, 2021-09-06:
Of course. The CSS is styled independently from the markup content.

R3dst0ne, 2021-09-06:
Changing to something simpler than HTML sounds like a good idea.
To be honest, every time I thought about changing something in the documentation, I got scared off. And I'm surely not the only one.

Markdown has the advantages that it is easy to read, edit and learn. And it can be converted to almost all other formats. The only disadvantage I can see is that basic markdown is very limited in some regards (E.g. positioning, tables, graphics). 
(These limitations can be overcome when HTML gets used, but that can easily defeat the purpose why we changed in the first place. Using a more advanced markdown version could do the same(but they come with their own set of problems).)

LaTeX has the advantages that content is easily readable, and its possibilities are cheer endless. It, too, can be converted/embedded in a lot of other formats. The only downside is that it is a little harder to learn. (IMHO, this is not a big disadvantage because most contributors come from the academic side, and there it is normal to know at least the basics.)

MarcinOrlowski, 2021-09-07:
I just discovered people use AsciiDoc. Got no experience with it, but maybe it is worth evaluating? 

https://en.wikipedia.org/wiki/AsciiDoc
https://asciidoc.org/

jxu, 2021-09-07:
From the Wikipedia page it looks like Markdown

MarcinOrlowski, 2021-09-07:
Syntax is similar but it can do more than ehat just plain markdown could. I've read that it has features useful for writing technical documentation too, so I'll take a look.

MarcinOrlowski, 2021-09-10:
I'd like to drive the separation more aggressively, so first thing we (I believe) already agreed on, is that separation can happen as long as we are able to generate downloadable form (PDF etc) of the documentation for off-line use (Java help is dead we do not need to care it really).

Second thing is what to use for  the documentation:
* [markdown](https://www.markdownguide.org/)
* [asciidoc](https://asciidoc.org/)
* [reStructuredText](https://www.sphinx-doc.org/en/master/usage/restructuredtext/basics.html)

all these can be easily converted to other formats and there are tools to generate PDFs etc from it. It's important however to clearly specify formatting features we must have. For now it looks like we need basic formatting (links, lists, tables, embedded images (do we require SVG support or just bitmaps?), preformatted text blocks.

Recent changes alter styling of some words so perhaps we would like to have ability to define i.e.o own tag and styling and then use single marker to have it applied (say we define `LSe` to be bold, with green bg and white text, then we can use `LSe` in some way to have this applied).

@mbaillif, as you spent most of the recent time editing docs, can you please post your thoughts on what would make maintaining docs easier?

@maehne @BFH-ktt1 @davidhutchens @zdimension

R3dst0ne, 2021-09-13:
In my understanding, we agreed that separation is a good thing, as long as we can have an offline version that ships with LSe.
If I remember correctly, the two options were either a PDF or offline webpages( that can be viewed from inside LSe).

What about [LaTeX](https://www.latex-project.org/)?
> Recent changes alter styling of some words so perhaps we would like to have ability to define i.e.o own tag and styling and then use single marker to have it applied (say we define LSe to be bold, with green bg and white text, then we can use LSe in some way to have this applied).

This and way more can easily be done with it.

MarcinOrlowski, 2021-09-13:
LaTeX is too complex for our needs and it will make contributing harder. I believe that we should be pretty OK even with plain markdown, but asciidoc or re can offer more features we might want to use to make docs look nicer.

Static HTML and PDF can be created easily from anything.

We should try to create one example page featuring everything that we may now need (headers, bold/italic, tables, embedded images, links), create it using all three and see what will be the outcome.

R3dst0ne, 2021-09-13:
> LaTeX is too complex for our needs and it will make contributing harder.

The setup is indeed harder, but editing/contributing is very easy to. The design has to be defined once and is then independent of the content, which is edited by the contributors. In some sense, it is even simpler. Its tags are more descriptive and things the other languages struggle with are very easy to implement.

LaTeX on an advanced level can be confusing to beginners, but these kinds of functions will rarely be used here. But the big advantage is that we can use them when needed.

An additional plus point is that LaTeX is widely known and used, and its support won't end in the foreseeable future.

BFH-ktt1, 2021-09-13:
> LaTeX is too complex for our needs and it will make contributing harder.

I might partially agree, but for me this is not a reason to exclude it. I think that we should keep it in the list of options

MarcinOrlowski, 2021-09-13:
As I said, the best (IMHO) approach now is to list all features we require, then create one page will all these features included using all the systems. Then we can see which one fits best and check PDF/HTML conversion tools. Then we decide.

jxu, 2021-09-13:
> An additional plus point is that LaTeX is widely known and used, and its support won't end in the foreseeable future.

Markdown is even easier to use and more well known. Github READMEs and this comment section use it.

R3dst0ne, 2021-09-13:
@jxu
You missed the point, this addition was not about ease of use and user base. 
Markdown only became well known over the last view years (maybe a decade). And this is reflected in its robustness. There are multiple competitors that implement nearly the same but have small differences so that they are not compatible. (What is considered the best changes yearly.) Markdown itself comes in a lot of different flavor because the original spec was poorly defined. This can lead to problems because there is no standardized way how things should be written/interpreted. A better explanation why this is a problem can be found [here](https://ericholscher.com/blog/2016/mar/15/dont-use-markdown-for-technical-docs/).
Meanwhile, LaTeX is going strong since decades.

maehne, 2021-10-02:
Sorry, I did just find now time to participate again in this discussion. I noticed that it has diverged a bit from the original question whether to separate documentation into a repository to in what format should be written the documentation in the future. Personally, I think that we should be first clear about our requirements to the user documentation before making the decision about the format and documentation framework to use. For me the multilingual aspect of our documentation/web site is important. It should be easy to add translations for individual pages, switching between them and ideally also keeping track of the translation status. Personally, I prefer HTML pages for online help over pointers into PDF documents. Web browsers are more optimized to the task of quickly displaying the information for a followed link and non-linear navigation. PDFs are more optimised for a linear reading flow. Page breaks may be annoying when reading PDFs online when splitting content, which should be presented together. It might be challenging implementing reliably and platform-independent the opening of specific links of a PDF file. Also, integration of multimedia content into PDFs is not well supported across all PDF viewers, because this relied, at least in the past, on Flash technology and Javascript. Personally, I would like to see that we build a nice web site for Logisim-evolution, which gives an overview about what it offers, provides download links to the built packages, contains the user and library documentation, some tutorials in either written form and/or videos, and links to any other useful ressources. Information for developers may be hosted there or here in the GitHub repository. The user and library documentation in form of a static website should be still part of the distributed installer packages of Logisim-evolution. I would prefer that it can be built offline as part of the general build process of Logisim-evolution. Fetching it from a remote site as part of the build may prove inconvenient for certain situations.

I completely agree that MarkDown is simple and easy to use, but has severe limitations when it comes to more complex documentation projects. Well, it was never intented to be used for that, but more for writing up shorter web content, e.g., for blog posts and other smaller text blobs, which explains its popularity and widespread use, e.g., also here on GitHub and similar web sites. ReStructuredText and AsciiDoc are certainly more up to the task of supporting writing extensive documentation, but besides the format itself, we should also have a closer look to the tooling ecosystem around them. Similar complexity and possibilities offer the various Wiki platforms with their different markup dialects.

In the past, I have been exposed to [DocBook](https://docbook.org/) and [DITA](https://www.xml.com/articles/2017/01/19/what-dita/) for writing complex technical (standard) documents. However, both are XML-based formats so that writing the content gets easily distracted by the many tags to be used and therefore requires support by a good dedicated editor. Also, the required XML tooling for transforming the documents to consumable HTML pages, PDFs, EPUBs, etc. tends to be comparatively complicated to set up and run.

Personally, I like LaTeX and have used it a lot over the past 22 years. It is very well suited for writing and *typesetting* structured documents, which get viewed as a PDF or printed on paper. In my experience, it is less suited for documents where content shall reflow dynamically, i.e., typically web pages and e-books. Though, I reckon that there are solutions offered by the LaTeX community on CTAN. However, you'll always need to restrict yourself to some subset of LaTeX to ensure that it will compile correctly to HTML/EPUB and PDF. What can be challenging is if you want to assemble parts of the content to different types of docuements (e.g., web site, online help, tutorials, etc.). Typically, you will also need to process everything to generate the new document. Integration of multimedia content in LaTeX documents remains challenging. The packages I am aware of either require Adobe Reader or some other special PDF viewer (e.g., `pdfpc`). This is good enough for giving presentations where you control the environment, but not for serving this to users on very different platforms.

To give an example how other open-source projects handle the documentation, I would like to point you to the [GHDL project](https://ghdl.github.io/ghdl). They have opted to use [Sphinx](https://www.sphinx-doc.org/en/master/) for all their documentation.