logisim-evolution/logisim-evolution #826: Duplicate pins in appearance

skywylder, 2021-07-16:
When I save and reopen a file in Logisim evolution, my circuits produce oscillation errors as the save has duplicated output pins and put them where my input pins are. This happens only in 2 of my subcircuits, those are LOAD_REGISTER and READ_RAM_BCD. I would attach my file here but GitHub won't let me. I have however attached a screenshot of the result.
[image]

MarcinOrlowski, 2021-07-16:
> I would attach my file here but GitHub won't let me.

Github allows ZIP archives.

skywylder, 2021-07-16:
> > I would attach my file here but GitHub won't let me.
> 
> Github allows ZIP archives.

[IGZ02e.circ.zip](https://github.com/logisim-evolution/logisim-evolution/files/6831109/IGZ02e.circ.zip)
How did I not think of this

maehne, 2021-07-22:
@skywylder: I had a look at your `.circ` file. Indeed, the symbol contains 4 input pins, which reference the very same input pin in the schematic. How this happened, I can only speculate. Your block `LOAD_REGISTER` is essentially empty except for input/output pins and do not connects:

[image]

The bug may have been triggered, because you placed the 1-bit output pin directly on top of the 1-bit input pin to connect them to each other. Logisim references pins in its `.circ` by there position and that may cause the problem you are observing.

I tried to reproduce your issue from scratch by creating a new Logisim design. On the main sheet, I only placed an input pin and output pin on top of each other:

[image]

When I try to save this design to a `.circ` file, it yields an *empty* file! Once, I pull them apart by one position, the correct XML seems to get generated. Once, I push the pin back to its original position and save, the `.circ` file is again empty. Additionally, after saving, I cannot interact anymore with the circuit. Only by clicking outside the schematic pane, e.g., selecting something in the library pane or by clicking on one of the icons on the top, the schematic pane becomes operational again. So, there's clearly a bug! @BFH-ktt1: Could you have a look please?

For the meantime, I suggest that you don't place pins on top of each other in schematics to not confuse Logisim. In the past, I also observed problems in some cases, where input and output pins were directly connected by a wire. Inserting a buffer fixed the problem in these cases.

BFH-ktt1, 2021-08-13:
This issue is more involved as seen in the first take.
The problem mentioned by @maehne is related to a null-pointer exception in the sorting algorithm of the xml-writer. This can be easily fixed. However, this is not a complete solution to the problem reported by @skywylder. It is even more complex, as loading a circuit A and writing it to a new file B (save as), than closing logisim, opening the circuit A again and saving it as file C would result in differences between A, B , and C, which should not happen. 
There are here multiple issues. I'm looking into the details

BFH-ktt1, 2021-08-14:
The problem really lies in the fact that the input is directly connected to the output, and the fact that a custom appearance is used. The custom appearance uses the position of the pins in the circuit to make a reference to the pins in the appearance. The problem here are:
1) Distinguish the correct input/output pin.
2) If multiple output pins are placed on top of each other (something that logisim allows) the problem becomes even worst.
I am trying to figure out a correct solution to these two problems which goes in following direction:
a) for problem 1) above: use the appearance shape to distinguish the correct pin, and make sure that the appearance description does not have more pins as the original circuit has.
b) for problem 2) preventing that you can place two outputs on the same location.

BFH-ktt1, 2021-08-14:
All the issues have been addressed in my latest PR. Closing the issue, please reopen if the problem persists.

closed 2021-08-14 by BFH-ktt1

MarcinOrlowski, 2021-08-14:
Can you please add link to mentioned PR here for a record?