logisim-evolution/logisim-evolution #827: Followup to nightly builds

davidhutchens, 2021-07-16:
I suggested @MarcinOrlowski use the environment variable JAVA_HOME_14_X64 to set the JAVA_HOME environment variable. From the Docs, that was the latest release on the system. But the Docs may be out of date. It is possible there is a newer version. On my Mac, the program /usr/libexec/java_home returns the location of the latest Java version. You could try that, or check to see if Java 15 or Java 16 is available by changing the above environment variable name. Or you could decide that the current choice or Java 14 is good enough. I just wanted to note there may be other options.

PR #819 changed the behavior of build.gradle.kts so it no longer correctly determines if some actions are needed on the Mac. This is a minor problem in that it fails by redoing an action rather than skipping a needed one. Nonetheless, it isn’t quite right. I’ll prepare a PR to fix it. While looking at it, I also noticed what looks like a small mistake in building the name of the *.deb file which may result in the same incorrect behavior. @MarcinOrlowski, please try the following on Linux: 1) run createDeb, 2) run createDeb again. The correct behavior of the second run is to report all actionable tasks up-to-date. If I am right about the mistake, it will rebuild the *.deb file. In that case, I’ll include that fix in my PR.

MarcinOrlowski, 2021-07-16:
Can we combine your efforts with #821 (Normalize package names), so we can eat cake and have cake? :)

davidhutchens, 2021-07-16:
Did you get a chance to test the createDeb build?

MarcinOrlowski, 2021-07-30:
Hm, I got a feeling nightly is not being run as expected. There's not much traffic in code base these days but I'd still expect a few builds anyway?

davidhutchens, 2021-08-13:
I am closing this issue since the bugs were fixed in #829 and @MarcinOrlowski has reopened #819 where he can address the suggestions in the first paragraph if he wishes.

closed 2021-08-13 by davidhutchens

MarcinOrlowski, 2021-08-13:
Sorry @davidhutchens  I missed that one (I got tons of GH notifications and miss one occasionally). I am going to check that as nightly is not working (IMHO), so that's good moment to see why. Will keep you posted.