logisim-evolution/logisim-evolution #830: Test vector from command line documentation

jxu, 2021-07-19:
In Logging > Test Vectors, it says to test with `logisim -test <circuitname> <vector.txt> <project.circ>`, but the command line options from `java -jar logisim-evolution.jar` show "-test name file   run test vector from a file against named circuit, then exit" and "-test-circuit <path_to_circ> open up a circ file and start the test bench within it. It retuns Success or fail."

jxu, 2021-07-19:
I've tried `java -jar logisim-evolution.jar -test main and_gate.circ and_gate.txt` and `java -jar logisim-evolution.jar -test and_gate.circ  and_gate.txt` but neither work. So the documentation is unclear.

[and_gate.zip](https://github.com/logisim-evolution/logisim-evolution/files/6843228/and_gate.zip)

R3dst0ne, 2021-07-20:
In `Command-line verification > Test Vectors` at the bottom it is described how to use it.
`java -jar logisim-evolution.jar -testvector <circuitname> <vector.txt> <project.circ>`

for your example, it would be:
`java -jar logisim-evolution-3.5.0-all.jar -testvector main and_gate.txt and_gate.circ`

jxu, 2021-07-21:
OK I found out I had mistakenly been using an old 2.14 jar

closed 2021-07-21 by jxu