logisim-evolution/logisim-evolution #831: Snap shapes to grid in circuit appearance

jxu, 2021-07-19:
Is there a way to get a fixed size rectangle for circuit appearance?

jxu, 2021-07-19:
From looking at the XML it appears that everything is integer grid snapped, which is much easier to control when zoomed in. But it would be nice if I could automatically make grid aligned shapes.

[image]

MarcinOrlowski, 2021-07-19:
Mind being more precise on what is desired result?

jxu, 2021-07-19:
Able to draw rectangles and lines with vertices snapping to the grid, like apparently for the ports

MarcinOrlowski, 2021-07-20:
So you mean being able to draw shapes on the circuit like lines etc, which are not part of circuit (i.e. no wires, no element shape), correct?

davidhutchens, 2021-07-20:
@jxu  is not drawing on the circuit. He is drawing on the circuit appearance. That is, he is drawing the way the circuit looks when placed in a higher level circuit. Drawing the rectangle in his picture above does not "snap to grid"; so the designer must carefully align it. I would do so at high zoom. The request that it optionally snap to grid is reasonable. Sometimes, it shouldn't be on the grid so it would need to be optional.

jxu, 2021-07-20:
That's correct. I'm talking about circuit appearance mode.

jxu, 2021-07-20:
I don't know any Java but it should be controlled here (code is named well but not commented)

https://github.com/logisim-evolution/logisim-evolution/blob/b698a76154c8f15e377056ee9237da160c9822fc/src/main/java/com/cburch/draw/tools/RectangularTool.java

jxu, 2021-07-20:
Looking through the code and reading the documentation carefully apparently this is already possible by holding down Ctrl.

closed 2021-07-20 by jxu