logisim-evolution/logisim-evolution #832: Copying/Pasting RAM, then moving it twice, causes it to disappear

trayres, 2021-07-20:
How to reproduce:
Select RAM item
Copy/Paste (Ctrl+C, Ctrl+V on Windows)
Move copied item once. While it is still selected, move it again.
The part will disappear, but selection handles will remain in its place. 
Now, you can click into where the RAM is supposed to be and, while it is selected, move it again.
Doing this over and over can create a bunch of handles for parts that aren't there. 

[image]

davidhutchens, 2021-07-21:
I could not duplicate this behavior on my Mac using our latest builds. Yes, I know you are using Windows. But It might help to narrow things down if you could give us the version of Logisim-evolution you are running.

trayres, 2021-07-21:
[image]

jxu, 2021-07-21:
Upload a MWE if possible

MarcinOrlowski, 2021-07-21:
> MWE

You meant [MCVE](https://stackoverflow.com/help/minimal-reproducible-example)?

jxu, 2021-07-21:
many acronyms for the same thing

MarcinOrlowski, 2021-07-21:
mind expanding your `MWE` then? :)

trayres, 2021-07-22:
I have not been able to reproduce this effect, let alone reproduce it in a MCVE.

I'll close the issue and re-open if I can duplicate the effect reliably.

closed 2021-07-22 by trayres