logisim-evolution/logisim-evolution #833: Can't build a pass transistor based latch

trayres, 2021-07-21:
[image]

This says oscillation, but it shouldn't register as an oscillation - it is a combinatorial loop, but it should stabilize. I'm not familiar with the codebase, can someone give me some pointers to look at the simulation engine?

trayres, 2021-07-21:
This works though (SR FF):
[image]

maehne, 2021-07-22:
Circuits with combinatorial loops, in which the signal values don't stabilise within the configured iteration limit, will trigger this "oscillation apparent" error. See the Logisim documentation under Logisim References -> Guide to Being a Logisim User -> Value propagation -> Oscillation errors. You will observe similar errors in other event-driven digital simulators when your combinatoric gates are modelled with zero-time propagation delay, which will cause the delta count at the current point in time to increase indefinitely until the simulator's set iteration limit.

This limits somewhat Logisim's capability to demonstrate the inner workings of some common logic components, which rely on such combinatorial feedback, as your Latch model demonstrates. However, these kind of circuits can typically better illustrated with a electronic-circuit-level simulation, e.g., done in LTspice. For more complex digital circuits, it is always recommended to use the memory components provided directly by Logisim, as these simulate more efficiently and more reliably.

As this limitation cannot be easily fixed due to the underlying simulation algorithm, I will mark this issue as won't fix. If you still would like to discuss/propose concrete improvements to the simulation engine, feel free to reopen this issue or start a new discussion/issue on the concrete topic.

closed 2021-07-22 by maehne

MarcinOrlowski, 2021-07-22:
Alternatively you may want to check [Digital](https://github.com/hneemann/Digital) instead, which uses different simulation approach and have this supported.