logisim-evolution/logisim-evolution #861: Incorrect data loaded into RAM/ROM

kokowadoko, 2021-08-09:
If loading a file into RAM or ROM that starts with any number of zero bytes followed by any number of non-zero bytes, the leading zero bytes will instead be replaced with uninitialized garbage data that is different each time you load it. This happens on both master and develop branches.

For example, if you load a file that looks like this:

    v3.0 hex bytes plain little-endian
    0000000067000000000000000000000000000000000000000000000000000000
    0000000000000000000000000000000000000000000000000000000000000000

Then the contents of the RAM/ROM will actually be something like this:

    v3.0 hex bytes plain little-endian
    df229dee67000000000000000000000000000000000000000000000000000000
    0000000000000000000000000000000000000000000000000000000000000000

Unless I'm missing something, this doesn't look like correct behavior.
The ROM configuration I discovered this on was 18 address bits and 8 data bits, but it seems to occur with any configuration and on RAM as well.
My machine is Arch Linux with OpenJDK 16. I haven't tested any other platforms.

kokowadoko, 2021-08-09:
Occurs on OpenJDK 11 as well.

davidhutchens, 2021-08-09:
I tried to reproduce this on my Mac with the current develop branch. Using the example file you gave, I consistently get the expected number of zeros loaded into ROM (or RAM) before the 67. I know you said you were using Arch Linux, but I would expect the same behavior on all platforms for this.

Could someone try to reproduce this in Linux?

kokowadoko, 2021-08-09:
I've encountered a few bugs since (editing ROM contents in the hex editor then clicking somewhere on another value corrupts a bunch of nearby values, and N-type transistors aren't propagating source to drain when the gate is high, and only when the source is high as well). Unless you can reproduce those bugs, then I'm starting to think there might be something wrong with my java installation. I'll see if I can figure anything out.

BFH-ktt1, 2021-08-09:
For the transistors please note that logisim is not an analog simulator, but a digital one, meaning that the transistors will only simulate the behavior of switches, and only if correctly connected. Hence please check your transistor circuit on correct usage.

kokowadoko, 2021-08-09:
I'm aware, but looking at the documentation it seems like this should work:
[image]
But instead the output is floating where it should be 1.

Interestingly, it works correctly when the source is 0:
[image]

Is this an incorrect configuration?
It works in original logisim, but not in logisim-evolution.

kokowadoko, 2021-08-09:
Oh, it seems like the behavior of the N-type transistor actually changed in logisim-evolution. I guess this isn't a bug then. Although I think there's a mistake in the docs.
[image]
The circled parts are in conflict with each other, I think.

The RAM/ROM corruption issue is still open though and I'm still trying to figure it out.

kokowadoko, 2021-08-09:
Ok, somehow I solved it by deleting ~/.java which contained the logisim preferences, so it was just a bad configuration all along.

Closing.

MarcinOrlowski, 2021-08-13:
I'd not close it that quickly. Invalid prefs content should be just dropped instead of affecting the app. Can we somehow re-instantiate the old contents to see what cause the confusion?

kokowadoko, 2021-08-14:
>Can we somehow re-instantiate the old contents

I don't think so. I don't have a backup of it.
The config was created years ago by original logisim, read (and maybe modified) by logisim-evolution more than a year ago, then fast forward to now testing out logisim-evolution again with the same config.
I tried to reproduce the config by deleting the config, running original logisim, running an older release of logisim-evolution from about a year ago, then running latest logisim-evolution, but I couldn't reproduce it.

mbaillif, 2021-08-14:
I note, I will correct the transistor doc problem.

hewzhew, 2026-09-02:
I retested the original sample on current `main` using the same 18-bit address / 8-bit data configuration and the exact `v3.0 hex bytes plain little-endian` contents. Addresses 0–3 remain zero, address 4 loads as `0x67`, and the following zero bytes remain intact. I also reran the complete 90-case `HexFileTest` suite successfully.

Since the original behavior disappeared after the reporter removed the old Java preferences and the problematic configuration could not be reproduced or preserved, I think this can be closed as no longer reproducible. If it occurs again on a current release, a new report with the memory file and the relevant preferences would give us something concrete to investigate.

davidhutchens, 2026-09-02:
I agree that there is nothing more we can do here as no problem can be reproduced. So I am closing this as resolved.

closed 2026-09-02 by davidhutchens