logisim-evolution/logisim-evolution #862: Logisim does not show connector pins

carbonmetrics, 2021-08-09:
I am trying out Logisim 3.5.0 Evolution on a Linux Mate 20.04 system (both the deb 64 version and the Java version).

The connections on the gates do not show up. 
I see on Youtube tutorials that they should be highlighted blue.
I can't connect my pins; they remain on `x1`.

How can I connect pins in Logisim evolution?

MarcinOrlowski, 2021-08-09:
Can you please please ZIP your example `.circ` file and attach to ticket?

carbonmetrics, 2021-08-10:
Thanks! File attached to ticket. 
The connectors show up sometimes, and then the whole program works. 
This is about 1 in 20 times when I start up the program. 
I have not been able to reproduce the behaviour.

R3dst0ne, 2021-08-10:
It seems 
>  File > Preferences... > Layout > Printer view

is sometimes enabled.
In this case this is not an issue with the pins but with saving/loading the preferences.

carbonmetrics, 2021-08-10:
Yes! That seems to solve the problem. I don't think I would have ever found this. Thanks!

BFH-ktt1, 2021-08-13:
Closing the issue

closed 2021-08-13 by BFH-ktt1