logisim-evolution/logisim-evolution #863: Shift Register documentation of pins is out-of-date

simoc, 2021-08-09:
Description of pins for Shift Register in online help under Help -> Library Reference does not match the pins displayed in a Shift Register component.

Pins are described along the north and south (top and bottom) of a Shift Register component, but logisim does not display pins in these positions.

R3dst0ne, 2021-08-10:
It currently only describes the `Classic Logisim` `Appearance`.
`Logisim-HolyCross` is the current default `Appearence`.
A description for the new `Appearence` is needed, but the `Classic Logisim` should not be removed.

davidhutchens, 2021-08-10:
There is a significant issue with different appearances in the documentation. RAM and ROM, for example, only show the Classic Logisim appearance and description. The Counter shows both pictures, but only describes Classic. The Register shows both pictures and describes a mix of them (sometimes accurate for one, sometimes for the other - see the enable or reset description).

We need a consistent,  clean, and simple way to describe the full behavior of all appearances. Then we should fix all documentation to match. Changing components documentation individually will likely lead to confusing, hard to read documentation. Perhaps we need a Discussion to design the change. Or perhaps documentation should be a point in the Roadmap discussion.

davidhutchens, 2021-08-12:
We do not intend to leave the documentation describing only the classic version. We appreciate your work in updating the documentation. But we really need the documentation to describe all the appearances in a clean, simple manner. That requires some careful thought.

MarcinOrlowski, 2021-08-13:
I think we need to drop legacy stuff. IMHO there's no point nor benefit from referring to legacy code or other forks (even if we get from them). All of these is dead for a long time. Maintaining legacy costs and I see no point of paying the rpice.

Let's discuss this as part of #695

maehne, 2021-08-25:
I agree partially with @MarcinOrlowski. I think our documentation efforts should focus on describing all components consistently using the default Logisim-evolution appearance. Supporting the old appearances is a service to our user base, which want still be able to open their old .circ files. Instead of offering the choice to change the appearance of each component, we may consider just offering the option to update a component instance to the new appearance. Maybe, also a kind of wizard could be implemented, which assists users in going through their .circ file finding all instances requiring update and allowing them to tweak manually the local wire routing after the change.

maehne, 2026-08-21:
Closing this issue as resolved by #864. Further enhancements to the documentation are always welcome.

closed 2026-08-21 by maehne