logisim-evolution/logisim-evolution #867: RISC-V IM simulator, blt instruction bug?

Antares95, 2021-08-11:
asm code:
```
_start:
        li a1, 1
        li a2, 2
        blt a1, a2, pc+0x100
        nop
```
does not jump to pc+0x100

logisim-evolution version is (commit 374a72318ecd0c16b7c077a2260753a1ffc2861c)

MarcinOrlowski, 2021-08-12:
Are you able to test using bleeding edge code (`develop` branch)?

Antares95, 2021-08-13:
> Are you able to test using bleeding edge code (`develop` branch)?

Yes, `develop` branch (version: commit 4a8dd3707afede432bbad9990b74e837248a5b81) seems to have same issue

BFH-ktt1, 2021-08-13:
This is possible, the Risc-V IM simulator was never tested in all details. Please also note that interrupt handling is not yet implemented for the Risc-V IM simulator. I will look into the details

BFH-ktt1, 2021-08-13:
Unfortanetely it was just a stupid typo in the assembler. Fixed in my latest PR. Closing

closed 2021-08-13 by BFH-ktt1