logisim-evolution/logisim-evolution #873: The controlled buffer two problems

mbaillif, 2021-08-14:
A) It does not have an IEC shape while the icons on the right exist.

B) Its output is (in my opinion) inconsistent, when its input is high impedance and its control is 1.

[image]

In the red square of the figure
I will make high impedance input -> output in error.

BFH-ktt1, 2021-08-14:
To A) I reported this already in #264. the problem is backward compatibility. As the controlled buffer in ANSI is very little, it is difficult to "cram in" the IEC symbol. Hence to do a proper job we have to drop backward compatibility on these components.
To B) I agree

mbaillif, 2021-08-15:
The #264 escaped me, 

I notice, that buffers and non have this symbol compatible in size and very similar.
Maybe a deviation from the IEC rules. I don't know.

Is it usable, acceptable ?

I imagine you have already thought of this, it's just to transcribe my thoughts. 
And there are other issues I don't know about.

mbaillif, 2021-08-22:
it is noted, I resume in al doc as soon as possible

BFH-ktt1, 2021-08-22:
@mbaillif : Please nite that the buffer has the same Problem B). I also corrected here the behavior. Thanks for updating the doc!

R3dst0ne, 2021-11-07:
@BFH-ktt1 @mbaillif 
Can this issue be closed?

closed 2021-11-07 by BFH-ktt1