logisim-evolution/logisim-evolution #884: Project icon

MarcinOrlowski, 2021-08-15:
I think we need something better than this:

[image]

Anyone with artistic skills, please contribute.

The only requirement for your contributions: it **MUST** be a SVG (vector) file. [Inskape](https://inkscape.org/)? [Krtita](https://krita.org/en/)? [Whatever](https://www.google.com/search?hl=en&q=free%20vector%20drawing%20app). Please contribute only SVG files though.

R3dst0ne, 2021-08-15:
> Some things you should change, some your should not. Since 2012 we stick with logisim-evolution, and it is something which is "burned into the mind". I'm for changes, but this one I would prefer not to change.

\- @BFH-ktt1 in [#721](https://github.com/logisim-evolution/logisim-evolution/issues/721#issuecomment-867922570)

IMHO, the same goes for the icon.

We could add a detail to make it easier to differentiate from the original Logisim.

@BFH-ktt1, @maehne
what do your think?

BFH-ktt1, 2021-08-19:
I agree to add some detail to it, but would stick with the design to acknowledge the work of Carl Burch

MarcinOrlowski, 2021-08-19:
IMHO there's absolutely no point of dragging the legacy just to drag the legacy **if** we can go better. Carl is acknowledged in the credits already.

MarcinOrlowski, 2021-08-21:
It is kinda tricky to come with nice icon that also stays readable when go low with the resolution (i.e. titlebar icon). So after some iterations I come with something like this:

[image]

Which just as simple as old icon, but different enough. Also it looks fine when scaled down to window icon:

[image]

or used in app menu:

[image]

Finally, it can be even combined with the current logo too:

[image]

R3dst0ne, 2021-08-24:
I'm against this change and propose to revert #904 until this is discussed further.
This is a change that should be part of the "roadmap discussion" #695 or at least be dissident by more than two people.

Maybe it is the nostalgia, after all I started using Logisim eleven years ago.
Or the fact that I'm not a  fan of the trend that icons and logos change more regularly.
But I don't see a reason to change the icon. Especially since, IMHO, the new icon is in no way better than the old one.

One other point against #904 are the problems mentioned in the [first comment](https://github.com/logisim-evolution/logisim-evolution/pull/904#issue-717283195) (`update_assets.sh` Linux exclusive, big `*.ico` file on Windows).

BFH-ktt1, 2021-08-25:
@R3dst0ne : It was never my intention to just "push" the new icon, but as there were no other people that reacted, and you did not commented to the new icon, I assumed (probably wrongly) that there were no objections. I'm sorry for that.

BFH-ktt1, 2021-08-25:
@MarcinOrlowski : Before I revert, do you still have the branch with this PR?

MarcinOrlowski, 2021-08-25:
Even if you merged prematurely, let's see the arguments against it we have here:
- Nostalgia :))) 
- "update_assets.sh Linux exclusive" it is helper script I wrote for myself and not part of functionality. Usage is optional.
- "big *.ico file on Windows)" it's most likely some odd issue with the tool used but I did not want to spend time on this as even the file is bigger, it's not a blocker. Anyone bothering can always find different tool or recreate `*.ico` using Gimp, PS or whatever else which would probably have the size back to original size.

And IMHO, from the all listed above, only the `*.ico` related one qualifies as valid concern from technical PoV.

Finally, as it is merged already, it'd require less work to discuss if it should be revoked, updated etc, instead of revoking, voting and then re-merging. For now it's 2 votes for vs 1 vote against.

MarcinOrlowski, 2021-08-25:
One note though: the reason I designed the logo and the changed the icon is project **branding**. LSe haven't had any branding
so far nor any visual representation. You could not do any pins, posters, flyers nor even basic website that looks less crappy than plain `man` page. With distinctive visuals you look and you know what it is, instantly, w/o reading. And so far we used icon from old, rusted, dead Logisim. Potential users of ours who went wrong path. There were 7456 downloads of old, abandoned Logisim on SF, just **this week**.

maehne, 2021-08-25:
Sorry for coming late to this discussion. I needed to prioritise other things after returning from vacation and due to the sheer amount of activity over the past weeks, it takes time to catch up. I appreciate your effort @MarcinOrlowski to improve the branding of Logisim-evolution. Already your new about screen improved the branding. However, I agree with @R3dst0ne that a broader discussion and maybe vote on your icon proposal should have taken place before merging to avoid later complains.

Personally, I liked the simplicity of the old icon and for me it was only a minor issue that it does not differ from the original one. In addition to the program icon, I would like to see a dedicated document icon for the `.circ` files. I agree that it is challenging to have an icon, which easily scales to different size while conserving recognisability/readability. To satisfy us nostalgic people, did you consider simply adding a red 'e' inside the OR gate of the original icon?

Anyway, I probably can get used to the new logo. However, its integration into the about screen, I like a bit less. The reason is that the uniformity of how "Logisim" was drawn is gone. A way, I see how to tune this would be to keep the PCB traces also for drawing the '0' letter, but make that letter circular and then integrate the new logo in the middle.

The new logo has an "inverted look", which seems to work better with dark desktop themes while the old one is clearly better suited for light desktop themes. So, maybe we should give the user two options: One suited for light desktop themes and one suited for dark ones. I don't consider this very urgent compared to other issues though.

R3dst0ne, 2021-08-31:
@maehne @BFH-ktt1 @davidhutchens @MarcinOrlowski @zdimension
We totaly forgot to discuss the "branding" topik.
How should we proceed?

maehne, 2021-09-01:
Personally, I would propose to revert the icon for the moment to the original one and decide about the new icon for version 3.7.0.

MarcinOrlowski, 2021-09-01:
I'd leave it as is mostly because it is already merged. There're two topics to be discussed, branding itself (which I believe we all share is good to have) and form of the branding. For now it's refreshed and mostly keeps on subject. And if at least one person asked "Which version of Logisim you use" and the answer will be like "green one", then we are home. Also as there's no million dollars in merch yet, we can change this at any time - and as it is already merged, I'd leave also to see the users reaction and possible feedback.

BFH-ktt1, 2022-08-26:
Can this issue be closed?

ianzone, 2025-01-06:
Hi, I hope the logo is less "bloody" 🙏🏼

maehne, 2025-01-17:
@ianzone: We are open to PRs improving the graphical assets in Logisim-evolution. This includes discussing proposals/drafts showing the direction of the rework of the graphical assets.

PR #904 (2021-08-24): New project icon
This PR adds new project icon and updated logo as discussed in #884. I also updates all PNG assets and introduces `artwork/update_assets.sh` shell script that automates generating PNGs from SVG sources. 

Unfortunately, the `update_assets.sh` is currently, (due to use of `icnsutils` and `icoutils`) Linux only (guess it may work on macOS but I haven't tried, too). We need to find reliable equivalent command line tools for these two packages to have this script converted into Gradle task and eventually have PNGs automatically generated on SVG update.

One minor issue with the sc [... cut]