logisim-evolution/logisim-evolution #890: Problem with the symbols "Floating point to Integer" and "Integer to Floating point"

mbaillif, 2021-08-15:
[image]

Icons and symbols of arithmetic components "Floating point to Integer" and "Integer to Floating point" are not very satisfactory. 

Test under windows 10   language FR-CH

simoc, 2021-08-16:
Maybe the text in the components would be correct if `\u2192` was used instead of `→` in the following Java source code:

FPToInt.java:

83: setIcon(new ArithmeticIcon("I→FP", 2));
103: painter.drawPort(OUT, "F→I", Direction.WEST);

IntToFP.java:

83: setIcon(new ArithmeticIcon("I→FP", 2));
102: painter.drawPort(OUT, "I→F", Direction.WEST);

MarcinOrlowski, 2021-08-25:
My comment from #892:

> I am still having feelings this only masks the problem instead of finding the culprit Do we know what was the real issue here in the first place? Anyone knows why exactly the issue happened? I am pretty surprised there's any issue with UTF-8 strings at all, as this is de facto encoding standard nowadays, and all our sources are expected to be UTF-8 encoded. So we need to understand what exact value we have in both cases and compare before we can tell we fixed anything :)

So unless we figure out what was the culprit really, please leave ticket opened despite merging #892

maehne, 2026-08-17:
@hewzhew: Sorry for the late feedback. The reason this issue was left open was that PR #892 was considered a workaround masking the real issue. @MarcinOrlowski's [comment above](https://github.com/logisim-evolution/logisim-evolution/issues/890#issuecomment-905434106) explained his view that using the arrow and other UNICODE characters directly in UTF-8 encoded strings should just work as well as replacing them with an explicit `\uxxxx` notation for the code point. So, we need to find these occasions, replace them with the character and check whether they are correctly rendered with the modern JDK versions we currently support. For details, please see the discussion in PR #892.

hewzhew, 2026-08-18:
Thanks for the clarification. I found that source encoding was explicitly fixed to UTF-8 for both `compileJava` and `compileTestJava` in #1284, a few months after #892. Current CI builds with Temurin 21, so this appears to remove the likely platform-default source-decoding condition behind the old workaround.

As a focused first step, I propose replacing the six `\u2192` display escapes in the three floating-point conversion components (`FpToInt`, `IntToFp`, and the newer `FpToFp`) with literal `→` characters. I would verify the icons and component labels on Windows/JDK 25, while the normal CI build covers the minimum supported JDK 21.

There are many other Unicode escapes in the Java sources, but they include control characters, parser character sets, and unrelated display symbols, so I would not replace them mechanically as part of this issue. Does this focused scope sound appropriate?

closed 2026-08-21 by BFH-ktt1