logisim-evolution/logisim-evolution #910: 74161 has improper behavior

z80jon, 2021-08-22:
This should be a fairly easy fix--the datasheet for the 74161 specifies that clearing is asynchronous--no clock pulse is needed, and enable/load line states are irrelevant. The moment nClear transitions low, all flip flops are set to 0 and the outputs reflect this change.

However, the 74161 in Logisim Evolution behaves more like a 74163--it only clears once a clock pulse comes in and nClear is low.

z80jon, 2021-08-22:
Link to relevant file: https://github.com/logisim-evolution/logisim-evolution/blob/374a72318ecd0c16b7c077a2260753a1ffc2861c/src/main/java/com/cburch/logisim/std/ttl/Ttl74161.java

Stand by--I think I'll make a pull request that should fix this. It's been a little while since I've done something like this.

Steps to reproduce: load any value into a 74161 (In my case: all enable pins and load are tied low), set clock to a low state, and flip nClear around. Nothing happens. If nClear is low when clock is pulsed, it does clear, but otherwise it does not.

z80jon, 2021-08-22:
Pull request: #911

maehne, 2021-08-25:
@z80jon: Thanks for opening this issue and also your willingness to create a PR to fix it. In the future, it would help if you could also point with an URL to the relevant data sheet (as long it is publicly and legally accessible).