logisim-evolution/logisim-evolution #925: Eclipse and development head

BFH-ktt1, 2021-08-25:
Logisim can be build by gradle, however, when importing the gradle project in Eclipse the "standard" compilation/run/debug will fail due to a missing dependency on ```com.cburch.logisim.generated.BuildInfo```.
This is kind-of annoying.
I might think of a dummy ```com.cburch.logisim.generated.BuildInfo``` that gets overwritten by a gradle-build.

MarcinOrlowski, 2021-08-28:
> Of course, as it is a generated file before doing anything in eclipse we have to do one time a gradle build

So it means `BuildInfo` is no longer up to date? Can we configure Eclipse builder to do something more (i.e. trigger anything) while building?

davidhutchens, 2021-08-28:
The problem is that Eclipse refuses to add a source path that doesn't exist at the time of the import to Eclipse. So running ./gradlew genBuildInfo before importing the project adds the generated files source path, but it isn't added otherwise. I have looked for a hook to get Eclipse to run a gradle tasks before import, but I haven't found it.

There are a couple of other problems I've encountered in playing with it (that I can fix). I may have a way to fairly easily update Eclipse from within Eclipse but I have more tests to run. I hope to get a full understanding and a PR soon.

MarcinOrlowski, 2021-08-28:
If that requires one manual build with Gradle before the import then this is acceptable (IMHO). What bothers me, is updating of BuildInfo class. Do I understand correctly that later, Eclipse will build using Gradle anyway? It's just matter of importing classes so Eclipse can i.e. index whatever, correct?

davidhutchens, 2021-08-28:
Eclipse doesn't use gradle to build.  It scans the build file and extracts the info for Eclipse's own build. Thus, it never builds the generated files for us. It does, however, make gradle tasks available once it is loaded so I can generate the file (and even the final packages) from within Eclipse. I'm looking at how I can generate the file and then tell Eclipse to update its info. If I can get that to work reliably and fairly easily, I'll document it.

The fall-back position is to require a command line build of the generated file(s) before importing into Eclipse.

I may add a genFiles task that depends on genBuildInfo and use that in the description. The idea is that we can then add additional files (such as the flex file, #563) without needing to change the documentation. What do you think?

MarcinOrlowski, 2021-08-28:
Let's do this: please do your tests and then update the developers docs. If we need build prior import - please add a note and invocation example to the docs. if we need to manually wire the task to make it auto-regenerated then great. Just make a note how. If not, then just add a note about rebuilding step is still manual for them. If they do care, because If they do not, then it is still fine.

MarcinOrlowski, 2021-08-28:
Can I assign this ticket to you @davidhutchens ?

davidhutchens, 2021-08-28:
Sure, assign me if you like. Your plan above matches what I was thinking.

MarcinOrlowski, 2021-08-28:
You can rebuild BuildInfo class file (w/o rebuilding whole project) you can use `genBuildInfo` Gradle task.

BFH-ktt1, 2021-08-29:
Resolved in #953 , closing

PR #953 (2021-08-29): Fix for Eclipse
This PR closes #925.

The srcDir change addresses @BFH-ktt1's specific problem in importing the project.

The change to git parameters in genBuildInfo addresses an issue that Eclipse doesn't have the correct current directory for the git command to work when Eclipse invokes the Gradle task.

This PR also adds a note for Eclipse users explaining what they need to do in order to successfully import and build the project.

Could @MarcinOrlowski and @BFH-ktt1 please review this.
closed 2021-08-29 by BFH-ktt1