logisim-evolution/logisim-evolution #956: Testers needed for Xilinx FPGA's

BFH-ktt1, 2021-08-29:
As the HDL-generation part has had a big overhaul, it would be great if also the Xilinx-FPGA path can be tested.
As I mainly use Intel (Altera) FPGA's, I do not have the ability for the moment to check the Xilinx Vivado and Xilinx ISE (outdated) design flow.
It would be great if someone that uses Xilinx FPGA's could give some feed-back and ideas (for example how to handle the situation when multiple boards are attached to the PC) on this design flow.

zdimension, 2021-08-30:
I've got somewere in a box a Virtex-II Pro PowerPC-based motherboard (Xilinx ML310) and a Virtex-6 PCIe development card (PLDA XpressV6), are those too old to be considered "testworthy" or should I try using Logisim with them?

BFH-ktt1, 2021-08-30:
@zdimension: As long as it has a button and a led it should be perfect. The main issue is to see if the vivado/ise command generators are working (hence a button connected to a led can be downloaded to the board from logisim). The HDL-generation is independent from the FPGA, so that is already tested.

maehne, 2021-08-31:
@zdimension: With your boards, you'll be able to test the Xilinx ISE flow. For Xilinx Vivado, we'll need users with more recent boards (Xilinx 7 series FPGAs or newer). The two boards, which you are listed, are not included in the list of known pre-configured boards. So you would have to create them in the FPGA Board Editor. Logisim-evolution knows the following popular boards with Xilinx FPGAs/SoCs:

- BASYS2 (ISE)
- BASYS3 (Vivado)
- DIGILENT_SPARTAN_3E_STARTER (ISE)
- ZYBO (Vivado)

@BFH-ktt1: By the way, I noticed that there's a board called DIGILENT_DE1_SOC. However, this board is made by Terasic (and Altera-based) and not sold by Digilent. Can we simply rename it?

BFH-ktt1, 2021-08-31:
@maehne : Done in my latest PR.

closed 2021-10-17 by BFH-ktt1