logisim-evolution/logisim-evolution #961: Map key to Button

R3dst0ne, 2021-08-29:
It would be useful if the ability, to map keys from the keyboard to buttons in the circuit, gets added.

PS. The doc of the button needs to be updated (#933).

hewzhew, 2026-08-20:
I investigated how this could fit the current input architecture. At present, `Canvas.MyListener` sends key events only to the selected tool, while `PokeTool` can send them only to the single component currently being poked. `Button.Poker` already contains the correct momentary active/passive state transitions, but only for mouse press and release.

A bounded, backward-compatible first slice could be:

1. add a project-wide opt-in setting for component keyboard mappings, defaulting to disabled;
2. add one optional persisted keystroke attribute to each Button, defaulting to none;
3. while the layout canvas has focus and the feature is enabled, route matching key press/release events to Buttons in the currently displayed circuit, reusing the existing momentary Button semantics; and
4. leave input Pins, multi-bit value cycling, window-global capture, and label-derived shortcuts for separate follow-ups.

This also needs a reusable persisted keystroke attribute/editor. The existing `JHotkeyInput` is coupled to application preferences, so it cannot be used directly as a component attribute editor.

Before implementing, could the maintainers please confirm three interaction choices?

- Is a Button-only first PR acceptable?
- Should duplicate mappings deliberately press all matching Buttons, or should duplicates be rejected?
- When the canvas has focus and the feature is enabled, should a mapped plain key take precedence over the selected tool's key handling?

My preference is to allow duplicate mappings to activate all matching Buttons and to consume a mapped key before tool handling. The project-wide opt-in keeps that behavior disabled for existing projects.

maehne, 2026-09-28:
Thanks @hewzhew for your investigation. I think a button-only first PR is perfectly acceptable with potential later extension to pins. For DIP switches, it might be difficult to define a meaningful semantic. To create less surprises, I would forward key strokes to buttons only when the respective Canvas and the Poke tool is active. IMHO, such a PR only makes sense if multiple key presses can be forwarded at the same time to buttons. The main reason, I see for this proposed enhancement, is that one cannot push several buttons at the same time using the Poke tool. It may make sense to limit the choice of keys primarily to letters and numbers. I would not allow duplicate mappings for the same key. Each button should get assigned a distinct key. It is probably OK to consume the mapped key before tool handling while the Poke tool is active.

I think that in the end a PR implementing this feature requires some extensive testing to see whether it creates some inadvertent new usability issues.

@BFH-ktt1 and @davidhutchens: Do you have any thoughts on this?

Vadimatorik, 2026-10-14:
Implemented for the Button component in #3090, following the constraints discussed above: one letter or digit per button, unique within the circuit, and delivered only while the canvas has focus and the Poke tool is selected, including several keys held at once.

DIP switches are not part of this change. If that behavior is needed, it should be a separate issue and pull request. Input pins are also left for a later change.