logisim-evolution/logisim-evolution #962: IEC and ANSI mixed up in DOC

R3dst0ne, 2021-08-29:
[image]

mbaillif, 2021-08-30:
I take note, and I correct later
Thanks

R3dst0ne, 2021-08-30:
It got mixed up in the text under the illustration as well.
> Because the `IEC` style tends to be more popular in the U.S., while the `ANSI` style tends to be more popular in Europe, some people refer to these styles according to these regions. 

(@mbaillif you did not correct this in c5405e2)

BFH-ktt1, 2021-08-30:
@mbaillif and @R3dst0ne :
Maybe I'm mistaking, but the "square-shaped" symbols are IEC (so to the right in the picture), and the left are ANSI-symbols.
Furthermore, IEC is used in Europe and ANSI in the rest of the world :-)

mbaillif, 2021-08-30:
The bigger it is, the less it shows :-))

I wanted to go too fast, I'm confused.

But I won't be caught again, I'll do it at the next PR.

BFH-ktt1, 2021-10-04:
Closing this issue as #916 has been merged

closed 2021-10-04 by BFH-ktt1