logisim-evolution/logisim-evolution #963: The Command-line options need some attention

R3dst0ne, 2021-08-29:
The Command-line options need some attention.
- `-analyze` #344
- `-accents` does not seem to work.
- `-testvector` is not mentioned by `-help` (outdated `-test` is)
- `-test-circuit`, `-test-fpga-implementation`, `-test-circ-gen`, `-questa` are undocumented.
- `-gates` I would prefer `ANSI` and `IEC` over `shaped` and `rectangular` for the gate style options.
- the windows executable does not support parameters-
...

IMO, it would be a good idea to conform to the [GNU Standards for Command Line Interfaces](https://www.gnu.org/prep/standards/html_node/Command_002dLine-Interfaces.html).

maehne, 2021-09-01:
I basically agree. However, we should only add short forms for options, where it makes sense and does not potentially cause confusion. Short options should be single letter, so that they can be combined if sensible.

MarcinOrlowski, 2021-09-01:
Unless you got different implementation, may I suggest the discussion to go in #981?

maehne, 2021-09-01:
Well, I will have my comments to your PR #981, but this issue is more general about the provided command line options and their documentation. We should make sure that your PR addresses all of them.

MarcinOrlowski, 2021-09-06:
@maehne any work pending here or we can have this ticket closed?

PR #983 (2021-09-03): Commons-cli based CLI args parser
This PR replaces old argument parsing with Apache's commons-cli based. 

* Completely rewritten command line argument parser:
    * All options have both short and long version now, 
    * All long arguments require `--` prefix i.e. `--version`,
    * All long arguments require single `-` as prefix i.e. `-v`,
    * `-clearprefs` is now `--clear-prefs` or `-cp`,
    * `-clearprops` option is remove (use `--clear-prefs` instead),
    * `-geom` is now `--geometry`,
    * `-nosplash` is now `--no-splash` or `-ns,
    * `-sub` is now `--substitute` or `-s`,
    * ` [... cut]
closed 2021-09-06 by R3dst0ne