logisim-evolution/logisim-evolution #966: LineBuffer strange behavior

BFH-ktt1, 2021-08-30:
In LedArrayGenericHDLGenerator.java, there is the line:
```
connections.add("{{assign}} s_{{ins}}{{id}}{{<}}{{pin}}{{>}} {{=}} {{1}};", info.getPinMap(pin).getHdlSignalName(info.getMapPin(pin)));
```

which throws an error:
```
WARNING: #E006: No mapping for placeholder: 'ins}}{{id}}{{<}}{{pin}}{{>'
Exception in thread "Thread-8" java.lang.RuntimeException: #E006: 1 unmapped placeholders detected in '{{assign}} s_{{ins}}{{id}}{{<}}{{pin}}{{>}} {{=}} s_LOGISIM_OUTPUT_BUBBLES[0];'. See console output for details.
	at com.cburch.logisim.util.LineBuffer.abort(LineBuffer.java:595)
	at com.cburch.logisim.util.LineBuffer.abort(LineBuffer.java:605)
	at com.cburch.logisim.util.LineBuffer.validateLine(LineBuffer.java:718)
	at com.cburch.logisim.util.LineBuffer.validateLineNoPositionals(LineBuffer.java:643)
	at com.cburch.logisim.util.LineBuffer.add(LineBuffer.java:293)
	at com.cburch.logisim.util.LineBuffer.add(LineBuffer.java:273)
	at com.cburch.logisim.fpga.hdlgenerator.LedArrayGenericHDLGeneratorFactory.getLedArrayConnections(LedArrayGenericHDLGeneratorFactory.java:270)
	at com.cburch.logisim.fpga.hdlgenerator.LedArrayGenericHDLGeneratorFactory.getArrayConnections(LedArrayGenericHDLGeneratorFactory.java:254)
	at com.cburch.logisim.fpga.hdlgenerator.ToplevelHDLGeneratorFactory.GetModuleFunctionality(ToplevelHDLGeneratorFactory.java:230)
	at com.cburch.logisim.fpga.hdlgenerator.AbstractHDLGeneratorFactory.GetArchitecture(AbstractHDLGeneratorFactory.java:411)
	at com.cburch.logisim.fpga.download.DownloadBase.writeHDL(DownloadBase.java:250)
	at com.cburch.logisim.fpga.download.Download.PrepareDownLoad(Download.java:403)
	at com.cburch.logisim.fpga.download.Download.run(Download.java:223)
	at java.base/java.lang.Thread.run(Thread.java:832)
```

MarcinOrlowski, 2021-08-30:
On it.

BTW: I noticed you renamed `openBracket` to `<` which is what I was thinking about too. Please ensure you replace all `bracketOpen` and `bracketClose` with short form and then remove support for `bracketOpen` and `bracketClose` form LineBuffer completely.

BFH-ktt1, 2021-08-30:
Fixed, closing

PR #968 (2021-08-30): Fixed handling of multiple placeholders in one line
Closes #966
closed 2021-08-30 by BFH-ktt1