logisim-evolution/logisim-evolution #969: Drop support for {{openBracket}} and {{closeBracket}} placeholders

MarcinOrlowski, 2021-08-30:
Currently LineBuffer's HDL placegolders offers `{{openBracket}}` and `{{closeBracket}}` and equivalent `{{<}}`/`{{>}}`. We shall keep only one, so unity usage across strings/text blocks and remove support for deprecated placeholder.

BLOCKED_BY #968 
BLOCKED_BY #952

R3dst0ne, 2021-08-30:
I'm in favor of the more descriptive placeholders (`{{openBracket}}`), the other ones could mean a lot of things.

MarcinOrlowski, 2021-08-30:
We already have `{{=}}` so actually `{{<}}` makes sense. And since it is in-code formatting (not user exposes), it is just fine to have it anything we decide works for majority. Another thing is that I personally hate exceptions (unless there's no other way). In this case we should choose just one form and keep using it, while all alternatives should be axed.

R3dst0ne, 2021-08-30:
I agree with you, consistency is important.
But I have a problem with the short placeholders:
- `{{<}}`/`{{>}}` could be comparison or bracket
- `{{=}}` could be comparison or assignment

Because of that, I'm in favor of more descriptive ones.

MarcinOrlowski, 2021-08-30:
> could be comparison or bracket

Do agree. You cannot tell from sign itself if it is un-equal comparision or bracket substitute. For now it is `bracket substitute`. Better approach?

>  could be comparison or assignment

Nope. Comparision is `==`.

BFH-ktt1, 2021-09-07:
Okay, I think that we have to drop ```{{openBracket}}``` and ```{{closeBracket}}``` for the reason that they are "long". As ```{{<}}``` and ```{{>}}``` are apparently dubious, what about:

1) ```{{(}}``` and ```{{)}}```

or

2) ```{{[}}``` and ```{{]}}```

maehne, 2021-09-07:
I agree that the spelled out form is probably too long and may render the templates less readable and that the short forms `{{<}}`,  `{{>}}` may be misinterpreted depending on the context. However, to make a good decision, could we please state here clearly in what context these placeholders are used and what semantics do they have? One aspect, we may also consider to take into account is how modern editors, who check for balanced parentheses react to the use of these placeholders.

MarcinOrlowski, 2021-09-07:
Actually I think `<`/`>` is best bracket we can use, simply because of readability. Please compare:

```
{{<}}  {{(}}  {{[}}
}}>}}  {{)}}  {{]}}
```

MarcinOrlowski, 2021-09-07:
> who check for balanced parentheses react to the use of these placeholders.

Wouldn't bother much.

MarcinOrlowski, 2021-09-07:
> However, to make a good decision, could we please state here clearly in what context these placeholders are used and what semantics do they have?

This is equvalent of `HDL.*()` calls you would have to otherwise do by hand for concatenation or placeholder values. Now this is built-in into LineBuffer, so in code you just need `addHdlPairs()` and you good to go.

```java
  public LineBuffer addHdlPairs() {
    return pair("assign", HDL.assignPreamble())
        .pair("=", HDL.assignOperator())
        .pair("or", HDL.orOperator())
        .pair("and", HDL.andOperator())
        .pair("not", HDL.notOperator())
        .pair("bracketOpen", HDL.BracketOpen())
        .pair("bracketClose", HDL.BracketClose())
        .pair("<", HDL.BracketOpen())
        .pair(">", HDL.BracketClose())
        .pair("0b", HDL.zeroBit())
        .pair("1b", HDL.oneBit());
  }
```

R3dst0ne, 2021-09-08:
Every thing has its downside.
`{{openBracket}}` is too long.
`{{{}}`, `{{(}}` and `{{[}}` are nearly unreadable.
`{{<}}` can be interpreted as something else.

If I'm not mistaken, those brackets get only used for splicing. So they could get removed as a hole and be replaced with a function like `splitVector()` as in #1057.
Overall
```java
"signal" + HDL.splitVector(num, 0)
```
is still shorter than
```java
"signal{{<}}" + num + HDL.vectorLoopId() + "0{{>}}"
"signal{{<}}{{num}}" + HDL.vectorLoopId() + "0{{>}}"
```
.
If something like `{{:}}` gets introduced for `vectorLoopId()`.
```java
"signal{{<}}" + num + "{{:}}0{{>}}"
"signal{{<}}{{num}}{{:}}0{{>}}"
```
it is still nearly the same length, but not as easily readable.

MarcinOrlowski, 2021-09-08:
For clarification - what `addHdlPairs()` adds is absolute minimum needed to "port" our HDL templates. It does NOT cover all the possibilities - only these needed for the conversion. Now, once @R3dst0ne added more helper methods the options are like this: 

1. shall `addHdlPairs()` add them all "just in case we needed it"?
2. shall we keep it as is and only add new pairs to `addHdlPairs()` when there's real demand.
3. I said meaning `{{<}}` is clear. Not any more. Once we had just brackets, it was clear, now it can (potentially) be "less than" type of comparision. So shall we replace `{{<}}` with something different and use `<` for "less than" (we already have `=` used)` or we shall replace i.e. `=` with `eq` so then `<` can be `lt` and `>` can be `gt` and `<=` can be `le` and `>=` can be `ge` (same as i.e. bash)?

> signal{{<}}{{num}}{{:}}0{{>}}"

It's hard to balance form and usability but I do not like the above proposal simply at first glance, because it is very cryptic if you look at it.

R3dst0ne, 2021-09-08:
> so then `<` can be `lt` and `>` can be `gt` and `<=` can be `le` and `>=` can be `ge`

Placeholders don't really work here because those operators need parameters (typecasting is needed).

>   > "signal{{<}}{{num}}{{:}}0{{>}}"
>
> It's hard to balance form and usability but I do not like the above proposal simply at first glance, because it is very cryptic if you look at it.

I did propose to use something like
`"signal" + HDL.splitVector(num, 0)`
instead of
`"signal{{<}}{{num}}{{:}}0{{>}}"`
because it nearly has the same length and is way easier to read and understand.

MarcinOrlowski, 2021-09-08:
> "signal" + HDL.splitVector(num, 0)

But please drop the concatenation:

```
"signal{{1}}", HDL.splitVector(num, 0)
```

maehne, 2021-09-09:
I think @R3dst0ne's arguments are sound. Dropping the concatenation as suggested by @MarcinOrlowski is fine with me, too.

PR #1053 (2021-09-13): Removed alternative HDL bracket placeholders
Closes #969

* #969
closed 2021-09-13 by MarcinOrlowski