logisim-evolution/logisim-evolution #971: Use Lombok codegen features

zdimension, 2021-08-30:
> "The shape of a program should reflect only the problem it needs to solve. Any other regularity in the code is a sign, to me at least, that I'm using abstractions that aren't powerful enough - often that I'm generating by hand the expansions of some macro that I need to write."

Java is a verbose language and prefers explicit code to syntactic sugar. Lombok provides annotations that allow generating boilerplate code for things such as member getters/setters.

It can work as a Gradle plugin and doesn't affect the build for the user.

Pros:
- less boilerplate

Cons:
- code may be less readable

Any yeas or nays?

zdimension, 2021-08-30:
ah there was already an issue

MarcinOrlowski, 2021-08-30:
There's PR adding lombok: #909 but we decided that that's not going to happen in this release. In fact we are now in code freeze stable and accept bug fixes only. Release by the end of the month and then we on track with `develop/` branch craze :)

MarcinOrlowski, 2021-08-30:
Reopening. On meeting #972 we discussed compiler plugin roughly. There is still some tension here but nonetheless, the not in the next release (which is scheduled by the end of September).

MarcinOrlowski, 2021-08-30:
There's PR #909 that shows impact of Lombok on our code base (TL;TR - LoC goes down).

There's another nice looking plugin called [manifold](http://manifold.systems/) but not yet sure about cost/benefit ration. License thing (it's not open source free project most likely). But if we can reduce LoC (esp. in the way that can be reverted really cheap)
it is still worth using it. The less LoC to maintain the better. And generative code usually is used to replace sometimes hundreds of lines of codes with just one annotation because we the all devs know what should be there. There's no point to explicitely talking
about that. It's **obvious** that it's there, so one annotation (like Lombok's `@Data`) instead of repeatable code is better.

MarcinOrlowski, 2021-09-06:
I am keeping updating the PR #909 with HEAD, so please let's have this discussed.

@davidhutchens @BFH-ktt1 @maehne

BFH-ktt1, 2021-09-06:
My remarks on PR #909:
```
Okay, had some time and did some test.
With the current version of lombok (1.18.20) it does definitely not work. You have to install the version 1.18.21 found in the link @davidhutchens shared above (projectlombok/lombok#2810). Than it compiles.
```
Hence I am not against lombok, but am also not jumping to get it in due to the current still "unstable" situation. We should when introducing this check each time if project lombok has updates, check with our tree and update our documentation.
Hence I wonder if the gains (a bit lesser boiler plate code) out weights the efforts it takes.

MarcinOrlowski, 2021-09-06:
> current still "unstable" situation

The "unstable" version is since I know this project. Yes, that's silly. Would they number it 2.17, you'd have no objections ;)

BFH-ktt1, 2021-09-06:
Nope, would the "normal install" work (hence as in their documentation) I would have lesser objections. All their official installers install 1.18.20. Only the fix in the indicated link installs 1.18.21 which announces itself as 1.18.20.

MarcinOrlowski, 2021-09-06:
I will update dev docs to mention that issue with Eclipse

MarcinOrlowski, 2021-09-08:
Ping. Can we make a decision? 

@BFH-ktt1 @davidhutchens

BFH-ktt1, 2021-09-13:
@davidhutchens : Could you also please give your take on this one?

davidhutchens, 2021-09-14:
Today, Java 17 GA was officially released. Java 17 is an LTS release. It will hit the normal distribution sites soon. That is relevant to this discussion since the evaluation of the risk associated with Lombok rests, in part, on how up-to-date they maintain their releases.

There are open issues and an open Lombok PR <https://github.com/projectlombok/lombok/pull/2936> related to Java 17. I grabbed Java 17 GA this morning and can now state that #1045 compiles with gradlew from the command line using Java 17 GA, which indicates these PRs are not yet using anything that Lombok is fixing.

However, I think it is wise to hold off a bit longer to see if they create a regular release that fixes our Eclipse issues and properly supports Java 17.

I have concerns about limiting the IDE choices of contributors. The major ones are covered by Lombok. As far as vi is concerned, it does not matter whether we use Lombok or not. Vi (as I use it) is not syntax aware anyway, and hence isn't really an IDE. Vim users may have more issues. I don't know.

I can't speak directly about Emacs, but there is Java capability and I did not see a Lombok addition. It is possible it isn't even needed. I think @maehne said he sometimes uses it and he may be able to tell us more by testing #1045 with Emacs to see how editing works.

MarcinOrlowski, 2021-09-15:
So I closed Lombok PR - let's see how the Eclipse support will proceed. I do agree that we may want to wait for them to stabilize the code. Let's see in a month or two and see if anything changed.

ze, 2021-10-05:
I saw this issue mention that Lombok PR that added support to Java 17 and after reading this discussion, has this team considered [immutables/immutables](https://github.com/immutables/immutables)? It has a different philosophy, but still seems to be an interesting library from what I've seen so far!

MarcinOrlowski, 2021-10-07:
We decided to not use any compiler plugins, at least for now. It may come back in future (and most likely will that way or another) but for now I am closing as 'wont-fix'

MarcinOrlowski, 2022-03-14:
Closing as Lombok did not attracted to much attention (politely speaking :)

closed 2022-03-14 by MarcinOrlowski