logisim-evolution/logisim-evolution #975: Problem with Xilinx boards in "Synthesize and Download"

R3dst0ne, 2021-08-30:
I found a problem while working on the HDL generation for the new PRNG.
This error occurs in `Synthesize and Download`. 
Up until `Map I/O resources onto board` everything works normal.
But after that, an exception is thrown when a board, that uses the Xilinx software, is selected. 
``` 
Exception in thread "Thread-1" java.lang.IllegalArgumentException: character to be escaped is missing
        at java.base/java.util.regex.Matcher.appendExpandedReplacement(Matcher.java:1019)
        at java.base/java.util.regex.Matcher.appendReplacement(Matcher.java:997)
        at java.base/java.util.regex.Matcher.replaceAll(Matcher.java:1181)
        at java.base/java.lang.String.replaceAll(String.java:2142)
        at com.cburch.logisim.util.LineBuffer.applyPairs(LineBuffer.java:349)
        at com.cburch.logisim.util.LineBuffer.add(LineBuffer.java:290)
        at com.cburch.logisim.util.LineBuffer.add(LineBuffer.java:273)
        at com.cburch.logisim.fpga.download.XilinxDownload.CreateDownloadScripts(XilinxDownload.java:223)
        at com.cburch.logisim.fpga.download.Download.CreateDownloadScripts(Download.java:203)
        at com.cburch.logisim.fpga.download.Download.PrepareDownLoad(Download.java:415)
        at com.cburch.logisim.fpga.download.Download.run(Download.java:223)
        at java.base/java.lang.Thread.run(Thread.java:831)
   ```
Reproduce:
- Create any circuit
- go to `Syntesize and Download`
- select a board that needs the Xilinx software
- click `execute` 
- complete `Map I/O resources onto board`

BFH-ktt1, 2021-08-31:
@R3dst0ne : Are you sure you use the head in development, as I cannot reproduce this error, the file that gives you the exception is generated correctly at my side.

R3dst0ne, 2021-08-31:
I just pulled the newest version ant this error still happens.
I created a circuit containing only one 1-bit input and a 1-bit output.
And used the `Annotate` function to label them.
(I have not selected the Xilinx Software and have no board connected)
I tested it with 3 Xilinx Boards(BASYS2, GECKO4LED, MICROLABTLE), and none of them works.

R3dst0ne, 2021-09-01:
I pulled again the newest develop version. 
It still doesn't work.
My Java version is:
> openjdk 16.0.1 2021-04-20
OpenJDK Runtime Environment AdoptOpenJDK-16.0.1+9 (build 16.0.1+9)
OpenJDK 64-Bit Server VM AdoptOpenJDK-16.0.1+9 (build 16.0.1+9, mixed mode, sharing)

BFH-ktt1, 2021-09-02:
@R3dst0ne : I downloaded the deb installer of the latest nightly build, installed it, and tried to reproduce the error without success, the file where you get the error (line 223 in XilinxDownload.java) gets created correctly without an error thrown. As this one is packaged with a JRE, it is not dependent on the one installed on my computer. I did not try MAC and Windows yet, will do that in a moment.

closed 2021-09-02 by BFH-ktt1