logisim-evolution/logisim-evolution #981: Command line arguments parser

MarcinOrlowski, 2021-08-31:
### Discussed in https://github.com/logisim-evolution/logisim-evolution/discussions/980

<div type='discussions-op-text'>

<sup>Originally posted by **BFH-ktt1** August 31, 2021</sup>
Currently the command-line parser in "Startup.java" is really "a mess" and not intelligent. It would be nice to modernize this parser such that it would also allow for long parameters like "--version" and their short ones "-v" (as proposed by @R3dstOne in #963).
Any ideas and any takers?</div>

MarcinOrlowski, 2021-08-31:
The side effect of using commons-cli is that we can have `--xxx` and `-x` options but not `-xxx` so that will be incompatible change. Still I'd go for it because `-xxx` is not standard approach anyway

maehne, 2021-08-31:
@MarcinOrlowski: IMHO, this is a duplicate of issue #963 and we should discuss your proposal there.

MarcinOrlowski, 2021-08-31:
#963 is related but not really a dupe. Also I already have args almost ported to commons CLI so I can take this one.

MarcinOrlowski, 2021-08-31:
#983 is early version of replacement CLI arg parsing code. Still WiP but PTAL

@BFH-ktt1 @R3dst0ne @maehne @davidhutchens

BFH-ktt1, 2021-09-04:
@MarcinOrlowski : can we close this one, as it is merged and we have other issues addressing the details?

PR #983 (2021-09-03): Commons-cli based CLI args parser
This PR replaces old argument parsing with Apache's commons-cli based. 

* Completely rewritten command line argument parser:
    * All options have both short and long version now, 
    * All long arguments require `--` prefix i.e. `--version`,
    * All long arguments require single `-` as prefix i.e. `-v`,
    * `-clearprefs` is now `--clear-prefs` or `-cp`,
    * `-clearprops` option is remove (use `--clear-prefs` instead),
    * `-geom` is now `--geometry`,
    * `-nosplash` is now `--no-splash` or `-ns,
    * `-sub` is now `--substitute` or `-s`,
    * ` [... cut]
closed 2021-09-04 by MarcinOrlowski