logisim-evolution/logisim-evolution #982: Circuit appearance is corrupted after save

yaybee, 2021-08-31:
In "Edit Circuit Appearance" of V3.5.0 you can change the order of the displayed items via right click options "Raise Selection" and "Lower Selection"
This changes the order of the displayed items and determines which items are overlapping (e.g. filled rectangles).
However during the "Save" operations the XML statements in the tag <appear> seem to be sorted alphabetically, whereby the required display order is corrupted.
Simply canceling the sort for the tag <appear> on save will solve this problem.
Note:  I found this bug while trying the construct the appearance of the build-in TTL circuits.

maehne, 2021-08-31:
Your description sounds like a duplicate of issue #501. It would be nice if you could build Logisim-evolution from the [develop branch](https://github.com/logisim-evolution/logisim-evolution/tree/develop) and test again. You can also download the latest nightly build from [here](https://github.com/logisim-evolution/logisim-evolution/actions/runs/1185404455).

MarcinOrlowski, 2021-08-31:
No need to build by hand - we have "nightly" build after each merge. Please see https://github.com/logisim-evolution/logisim-evolution/tree/develop#nightly-builds

yaybee, 2021-08-31:
The latest nightly build .msi will not start on my system, so I cannot test it at the moment. I had a look at #501 and #708 and it seems the same. I will try to test it later.

maehne, 2021-08-31:
Could you give the [JAR](https://github.com/logisim-evolution/logisim-evolution/suites/3639551865/artifacts/88193938) a try?

MarcinOrlowski, 2021-08-31:
What is the problem with MSI installer?

BFH-ktt1, 2021-08-31:
This has been fixed in the current head of develop and hence in V3.6.0 that will be released soon,

yaybee, 2021-09-02:
[image]

I reinstalled the nightly and it now works OK. Probably remove of old 3.5.0 was not done completely.
The tags within "appear" tag are no longer sorted, except for the "circ-port" (and "circ-anchor") tags. These always end-up at the top so the look of the build-in TTL with half hidden ports cannot be created. See attached jpg with a TTL 7400 and a homemade 4026.

BFH-ktt1, 2021-09-02:
@yaybee : This is correct, and not related to the sorting, but to how logisim does the drawing of the ports, this is done after the shape is drawn. Here there is a choice and as ports could be completely hidden by a shape if we do draw the ports before drawing the shapes I opt for a "won't fix" label.

yaybee, 2021-09-02:
No problem, my issue was already solved, but I could not find it in the issues. In future I will try the nightly first. Thanks for the support.

BFH-ktt1, 2021-09-02:
Your're welcome, and thanks for your issue. Closing the issue, feel free to reopen it something is still wrong.

closed 2021-09-02 by BFH-ktt1