logisim-evolution/logisim-evolution #987: Move TTL, TCL, BHF, I/O Extra, and SOC Libraries to Independent JARs

TChapman500, 2021-08-31:
While it's good to have extra libraries for Logisim, I feel that these libraries should be as independent JAR libraries rather than built-in Libraries.  This would allow such libraries to be used with almost any version of Logisim or Logisim Evolution and would reduce the workload of resolving pull conflicts.

MarcinOrlowski, 2021-08-31:
In general it sounds like a good idea but I am against this type of changes as it brings no real benefit and is expensive to make and maintain in future. Also it may be more problematic to handle by the users. 

Voting to close as `wont_fix`.

MarcinOrlowski, 2021-08-31:
Form me, there is no benefit from doing this change. It will improve nothing really. Also I also do not really buy the argument about "reduce the workload of resolving pull conflicts." - this is not a problem here. Again, it solves nothing.

BFH-ktt1, 2021-09-01:
@TChapman500 : I do not quite understand ```would reduce the workload of resolving pull conflicts```. Up to this moment I did not see any problems due to the libraries concerning pull conflicts. Also the libraries can be made "invisible" to the user by modifying the standard template. Moving the first four to a jar library should not pose any problem (also because they are quite "static" for some time now), however the SOC library has some "roots" inside the simulation engine of logisim, and might be quite difficult to move to a separate jar.
Seen that the reasons for moving it is mainly the above mentioned line, I do not see a profound reason to move those libraries and would also vote for ```wont_fix```

maehne, 2021-09-01:
@TChapman500: Please, be more specific *where* you observe pull conflicts, which cause additional workload. Do you mean pull conflicts in Logisim-evolution's source code itself, in circuit files, or somewhere else? Maybe, it is best if you describe your use case where you observe the pull conflicts.

TChapman500, 2021-09-01:
These conflicts would be between forks that don't have these libraries (eg: Holy Cross) and this repository.  Also note that there are some differences between the additional components of the I/O library between different forks.

maehne, 2021-09-01:
We cannot and probably should not strive for compatibility among the diverse Logisim forks. There developments have already diverged too much even though we have tried to port back features from other forks to Logisim-evolution. However, maybe it would be feasible to only require the libraries in the .circ files, which get actually used in schematics by instantiating a component. What do you think @BFH-ktt1?

TChapman500, 2021-09-01:
I could go for that.

BFH-ktt1, 2021-09-01:
@maehne : Your suggestion would not solve the problem, as the libraries would still be in the "source tree", hence pulling would still pose the same problems as mentioned by @TChapman500. I would rather go for a solution where we try as much as possible to merge back the forks into evolution and than bundle resources to work on one version where everybody is happy with.

R3dst0ne, 2021-09-01:
The same topings get discussed in #690.

MarcinOrlowski, 2021-09-07:
Closing as `wont_fix`

closed 2021-09-07 by MarcinOrlowski