logisim-evolution/logisim-evolution #988: Slider Bug:  64-Bit Outputs Restrict Slider to Half of Max Range

TChapman500, 2021-08-31:


MarcinOrlowski, 2021-08-31:
Please provide reproduction steps?

TChapman500, 2021-08-31:
Just put a slider onto the circuit, set the bit width to 64 bits, and then try to move the slider all the way to the right.  It'll stop at exactly half way `0x7FFFFFFFFFFFFFFF`.

BFH-ktt1, 2021-09-01:
I have been spending the last hours to "fix" the bug and rewrite the slider source (it was kind of a mess), to come to the conclusion that actually the current implementation of the slider makes no sense, rationale:
1)  The number of slider positions is fixed to 100 positions, so taking more that 7 bits for the output makes no sense as only the most significant 7-bits of the value would change.
2) The value attribute is also questionable, as this is the same functionality as a pin.
3) To get a similar behavior on more than 7 bits a multiplier could be used.

I therefore opt for a ```won't fix``` label and replace the current slider implementation where you can select the "number of slider positions" in a 2^n fashion. In this case the nr. of output bits would be n, where n should be restricted to f.e. 1 <= n <= 8. the width of the slider would then be 2^8 pixels on the screen.

Do I make any sense?

TChapman500, 2021-09-01:
That does make sense.  Though I would like to request a maximum of 10 bits, simply because that's how many bits the ADC on the Arduino gives you.  That would result in a sliding region of up to 1024 pixels (not counting the buffer area around the slider).

BFH-ktt1, 2021-09-01:
I'm not against, but take note that 1024 pixels makes a very "big" component. On a screen of 1280x1024 (I know this is a low res) you would not be able to see the whole components, hence poking would be partially impossible. A resolution of 10-bits would also be possible with 2 sliders of each 5 bits.
What do you think?

MarcinOrlowski, 2021-09-01:
With such huge values it pointless to offer so high precision with the slider. We should have stlider to move by groups (say. of 10/25/50, whatever) and separate buttons"+" and "-" to move by 1. That would allow to keep size at bay.

MarcinOrlowski, 2021-09-01:
@BFH-ktt1 while you will be working on the slider - can you please also fix so Hex Radix will be default and that all radix can have prefix/suffix for easy reading (i.e. for hex let's go `0x??????` for binary add "b" at the end for octal add "o" as prefix. For signed we should show "+"/"-" (with exception of 0), for float small "f" as suffix?

BFH-ktt1, 2021-09-02:
@TChapman500 : I reworked the slider in my latest PR. The number of bits (slider positions) can be selected in the top of the file. I put for the moment 8-bits as maximum, as this gives a reasonable size, but am open for more. Closing the issue.

closed 2021-09-02 by BFH-ktt1