logisim-evolution/logisim-evolution #990: Pick tick frequency in `.circ`

donno2048, 2021-09-01:


maehne, 2021-09-01:
@donno2048: Thanks for your suggestion, but please be more specific with your feature request and justify it!

donno2048, 2021-09-01:
Ok, I think it would be great to have the option to choose the tick frequency in the `.circ` file; if I shared my project, I would like the person who gets it to see it at the same speed I intended and not the frequency he already set-up in _logisim-evolution_

maehne, 2021-09-01:
@donno2048: Thanks for this clarification. It makes sense.

BFH-ktt1, 2021-09-02:
@donno2048 , @maehne : Would it not make more sense to store the tick-frequency in the "clock" component, as "normally" only this component is affected by the clock frequency

maehne, 2021-09-02:
@BFH-ktt1: Good question. Personally, I always set the tick frequency via the simulation menu and not via the clock component. Depending on your project, it might be wishful that the speed of the interactive simulation can be configured per circuit sheet, as one might check things on different time scales in different parts of the project.

However, we should probably distinguish two tick frequencies per circuit sheet (or maybe clock component): The tick frequency for interactive simulation and the tick frequency for downloading to an FPGA. This would also make our live easier reviewing the Logisim projects, which we do each spring term.

BFH-ktt1, 2021-09-03:
@maehne : makes sense, I'll think a bit about it, and modify my PR accordingly.

BFH-ktt1, 2021-09-04:
@maehne , @donno2048 : The option is now in the development branch. Each circuit will save/restore the last used tick-frequency and download-frequency. #1020 is a PR to also integrate it in the up coming v3.6.0, but due to feature freeze it might not make it in there.

BFH-ktt1, 2021-09-04:
It has been merged into develop and the up comming 3.6.0 version. Closing the issue, please reopen if something is missing

closed 2021-09-04 by BFH-ktt1