logisim-evolution/logisim-evolution #992: Update source code copyrights header

MarcinOrlowski, 2021-09-01:
I am going to sync source code copyright header with the project + update copyright in-app scroller to match.

This means:
* I will shorten the copyright header
* I will remove list of contributors from source header
* I will sync in-app scroller content with [credits page](https://github.com/logisim-evolution/logisim-evolution/blob/develop/docs/credits.md)

Please:
* check if [credits page](https://github.com/logisim-evolution/logisim-evolution/blob/develop/docs/credits.md) needs updating
* other comments?

MarcinOrlowski, 2021-09-01:
Proposed new header:

```
/*
 * Logisim-evolution - digital logic design tool and simulator
 * Logisim-evolution Copyright ©2001-2021 Logism-evolution developers
 * 
 * https://github.com/logisim-evolution/
 * 
 * This is free software released under GNU GPLv3 license
 */
```

Current source header

```
/*
 * This file is part of Logisim-evolution
 *
 * Logisim-evolution is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by the
 * Free Software Foundation, either version 3 of the License, or (at your
 * option) any later version.
 *
 * Logisim-evolution is distributed in the hope that it will be useful, but
 * WITHOUT ANY WARRANTY; without even the implied warranty of MERCHANTABILITY
 * or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU General Public License
 * for more details.
 *
 * You should have received a copy of the GNU General Public License along
 * with logisim-evolution. If not, see <http://www.gnu.org/licenses/>.
 *
 * Original code by Carl Burch (http://www.cburch.com), 2011.
 * Subsequent modifications by:
 *   + College of the Holy Cross
 *     http://www.holycross.edu
 *   + Haute École Spécialisée Bernoise/Berner Fachhochschule
 *     http://www.bfh.ch
 *   + Haute École du paysage, d'ingénierie et d'architecture de Genève
 *     http://hepia.hesge.ch/
 *   + Haute École d'Ingénierie et de Gestion du Canton de Vaud
 *     http://www.heig-vd.ch/
 */
```

@BFH-ktt1 @maehne @davidhutchens

BFH-ktt1, 2021-09-02:
This line:

```Logisim-evolution Copyright ©2001-2021 Logism-evolution developers```

will be a head age to maintain (meaning that each year we have to update all headers).

For the rest I'm fine with shortening the headers.

MarcinOrlowski, 2021-09-02:
I think we can have years removed w/o no probems. Something like this:

```
/*
 * Logisim-evolution - digital logic design tool and simulator
 * Logisim-evolution Copyright by Logism-evolution developers
 * 
 * https://github.com/logisim-evolution/
 * 
 * This is free software released under GNU GPLv3 license
 */
```

BFH-ktt1, 2021-09-04:
This is merged, hence closing issue

PR #1005 (2021-09-03): Updated source copyright header
No other changes but mass search/replace + a few added headers extra.

Because every single file was touched, Checkstyle will run on all of them too, failing miserably. This is expected and should be ignored.

Partially fixes #992
closed 2021-09-04 by BFH-ktt1