logisim-evolution/logisim-evolution #1010: Command line options are not correctly parsed after #983

BFH-ktt1, 2021-09-03:
All command line options work, but their parameters are not (always) correctly parsed and return a null pointer, examples:
```
java -jar build/libs/logisim-evolution-3.7.0-all.jar -g shaped
Exception in thread "main" java.lang.NullPointerException: Cannot invoke "String.equals(Object)" because "gateShape" is null
	at com.cburch.logisim.gui.start.Startup.handleArgGates(Startup.java:520)
	at com.cburch.logisim.gui.start.Startup.parseArgs(Startup.java:380)
	at com.cburch.logisim.Main.main(Main.java:74)
```
or
```
java -jar build/libs/logisim-evolution-3.7.0-all.jar -f ~/leguantest.circ main LEGUAN
Exception in thread "main" java.lang.NullPointerException: Cannot read the array length because "optArgs" is null
	at com.cburch.logisim.gui.start.Startup.handleArgTestFpga(Startup.java:645)
	at com.cburch.logisim.gui.start.Startup.parseArgs(Startup.java:386)
	at com.cburch.logisim.Main.main(Main.java:74)
```

BFH-ktt1, 2021-09-03:
@MarcinOrlowski : PR #1011 did not fix:
```
java -jar build/libs/logisim-evolution-3.7.0-all.jar -f ~/leguantest.circ main LEGUAN
[main] ERROR com.cburch.logisim.gui.start.Startup - Invalid arguments for --test-fpga option. Supported are: <circ_input> <circuit_name> <board> [tick frequency] [HDLONLY]"
```
the option given on the command line are correct

BFH-ktt1, 2021-09-03:
@MarcinOrlowski : Also the help screen is now a bit "bizarre" as it seems to be limited in the number of characters each line:
```
Logisim-evolution v3.7.0
https://github.com/logisim-evolution/
commandline/31c52a93 (2021-09-03T11:11:40+0200)
Java HotSpot(TM) 64-Bit Server VM v16.0.2 (Oracle Corporation)

usage: Logisim-evolution [-b <file>] [-e] [-f] [-g <shape>] [-h] [-j
       <args>] [-l <file>] [-m <geometry>] [-n] [-o <lang>] [-p] [-r] [-s
       <file>] [-t <format>] [-v] [-w <argTestVectorOptionArgName>] [-z
       <file>]
 -b,--test-circuit <file>                        Open up a circ file and
                                                 start the test bench
                                                 within it. It returns
                                                 Success or fail.
                                                 Arguments: <path_to_circ>
 -e,--empty                                      Use empty template.
 -f,--test-fpga                                  Test implementations
                                                 design using circ_input
                                                 file and optional
                                                 map_input mapping file
                                                 (obsolete since v3.3.3).
                                                 The implementation
                                                 circuit_name is the
                                                 circuit choose to
                                                 implement (FPGA_Top) and
                                                 the board is the board
                                                 name located in
                                                 resources/logisim/boards/
                                                 without the xml
                                                 extension. The optional
                                                 tick frequency is in Hz.
                                                 The optional switch
                                                 HDLONLY will only
                                                 generate the
                                                 HDL-description and will
                                                 not perform synthesis,
                                                 P&R, and Download.
                                                 Arguments: <circ_input>
                                                 [map_input]
                                                 <circuit_name> <board>
                                                 [tick frequency]
                                                 [HDLONLY]
 -g,--gates <shape>                              Sets "shaped" or
                                                 "rectangular" gate style.
 -h,--help                                       Displays this argument
                                                 summary help page.
 -j,--test-circ-gen <args>                       Open up the circ file and
                                                 write a new one to later
                                                 check the XML content.
                                                 Arguments: <circ_input>
                                                 <circ_output>
 -l,--load <file>                                Load image file into RAM
                                                 (works with -tty only).
 -m,--geometry <geometry>                        Sets geometry for main
                                                 window: Arguments:
                                                 WIDTHxHEIGHT[+X+Y]
 -n,--no-splash                                  Hides splash screen at
                                                 startup.
 -o,--locale <lang>                              Sets locale as given as
                                                 argument.
 -p,--plain                                      Use standard Logisim
                                                 template.
 -r,--clear-prefs                                Clear application
                                                 preferences at startup.
 -s,--substitute <file>                          Substitutes library lib1
                                                 with lib2. Arguments: lib
                                                 lib2.
 -t,--tty <format>                               Run without graphical
                                                 interface. Argument: TTY
                                                 format.
 -v,--version                                    Display version number
                                                 and exit
 -w,--test-vector <argTestVectorOptionArgName>   argTestVectorOption
 -z,--template <file>                            Use file as template.
                                                 Argument: file.
```

MarcinOrlowski, 2021-09-03:
This class sets line width to 74. Unfortunately it do not adapt to terminal width. I can now just set it wider (100?) and that's it because I do not think this is a serious issue worth further works to figure out proper way of getting the terminal width in crossplaftorm way,

BFH-ktt1, 2021-09-03:
Works for me

MarcinOrlowski, 2021-09-06:
@BFH-ktt1 can this ticket be closed now?

BFH-ktt1, 2021-09-06:
Nope, the 

```java -jar build/libs/logisim-evolution-3.7.0-all.jar -f ~/leguantest.circ main LEGUAN
[main] ERROR com.cburch.logisim.gui.start.Startup - Invalid arguments for --test-fpga option. Supported are: <circ_input> <circuit_name> <board> [tick frequency] [HDLONLY]"
```

still does not work, the -f option does not seem to accept false or correct input, it always throws an error

BFH-ktt1, 2021-09-08:
There are still two small problems:

1) The correct command-line options are not displayed:
```
java -jar build/libs/logisim-evolution-3.7.0dev-all.jar -f
[main] ERROR com.cburch.logisim.gui.start.Startup - Failed processing command line arguments.
```

2) The help text still contains the [map_input] entry:
```
 -f,--test-fpga <arg>                            Test implementations design using circ_input file.
                                                 The implementation circuit_name is the circuit
                                                 choose to implement (FPGA_Top) and the board is the
                                                 board name located in resources/logisim/boards/
                                                 without the xml extension. The optional tick
                                                 frequency is in Hz. The optional switch HDLONLY
                                                 will only generate the HDL-description and will not
                                                 perform synthesis, P&R, and Download. Arguments:
                                                 <circ_input> [map_input] <circuit_name> <board>
                                                 [tick frequency] [HDLONLY]
```

3) When providing a wrong argument the command get's executed, which it should not:
```
java -jar build/libs/logisim-evolution-3.7.0dev-all.jar -f ~/leguantest.circ main LEGUAN 200 HDLONLY$
[main] ERROR com.cburch.logisim.gui.start.Startup - Invalid arguments for --test-fpga option. Supported are: <circ_input> <circuit_name> <board> [tick frequency] [HDLONLY]"
[main] INFO com.cburch.logisim.fpga.gui.Reporter - Building netlist for sheet "main"
[main] INFO com.cburch.logisim.fpga.gui.Reporter - Circuit "main" has 4 nets and 1 buses.
[main] INFO com.cburch.logisim.fpga.gui.Reporter - Circuit "main" passed DRC check.
[main] INFO com.cburch.logisim.fpga.gui.Reporter - The Board LEGUAN has:
[main] INFO com.cburch.logisim.fpga.gui.Reporter - 36 Pin(s)
[main] INFO com.cburch.logisim.fpga.gui.Reporter - 1 LEDArray(s)
[main] INFO com.cburch.logisim.fpga.gui.Reporter - 12 Button(s)
[main] INFO com.cburch.logisim.fpga.gui.Reporter - 2 DIPSwitch(s)
[main] INFO com.cburch.logisim.fpga.gui.Reporter - 4 SevenSegment(s)
...
```

reopening
```

BFH-ktt1, 2021-09-08:
Unfortunately still not completely fixed:
```
java -jar build/libs/logisim-evolution-3.7.0dev-all.jar -f ~/leguantest.circ main LEGUAN 200 HDLONLY$
[main] ERROR com.cburch.logisim.gui.start.Startup - Invalid tick frequency: HDLONLY$
```
This has nothing to do with the tick-frequency :-)

and point 1) above did not change

MarcinOrlowski, 2021-09-08:
It does not but as 4th and 5th elements can be literally swapped, then code checks if value is literal `HDLONLY` (it is not, because of `$`, then it tries to parse int to get tick freq. This step fails for obvious reason, hence the message. I'd leave as is.

MarcinOrlowski, 2021-09-08:
or I know how I can easily  improve this. Will send new PR

PR #1011 (2021-09-03): Fixes CLI argument parser config
Fixes CLI argument parser config that was incorrect for certain options.

Closes #1010
PR #1055 (2021-09-07): Fixed handling of --test-fpga option arguments
Closes #1010 

* #1010
PR #1058 (2021-09-08): Fixes --test-fpga not aborting when handling invalid arguments 
Closes #1010 

* #1010
PR #1059 (2021-09-08): Improved --test-fpga arguments handling
The `--test-fpga` argument handler should be bit smarter now.

Closes #1010 

* #1010
closed 2021-09-13 by BFH-ktt1