logisim-evolution/logisim-evolution #1012: Nightly fails on MacOS build

BFH-ktt1, 2021-09-03:
I get more often an error:
```
Nightly: Some jobs were not successful 

Nightly / macOS build
Failed in 14 seconds
```

MarcinOrlowski, 2021-09-03:
It's seems that is about lack of JDK 16 by default. Needs to check if we need different env var or just install it. @davidhutchens could you please take a look?

davidhutchens, 2021-09-03:
I looked at the output from the successful and failing macOS builds. My best understanding is that OpenJDK 16.0.2 is actually the default installed Java. It appears to have been set before we set JAVA_HOME. My guess is that older ones are available through explicit setting of JAVA_HOME, but they did not provide a variable for the latest release.

I suggest removing the line to set JAVA_HOME and see if that works.

MarcinOrlowski, 2021-09-04:
@davidhutchens Sure, will give it a try.

BFH-ktt1, 2021-09-04:
Has been solved with #1037. Closing issue

PR #1037 (2021-09-04): Removed explicit JAVA_HOME setup on macOS nighty builder
Closes #1012
closed 2021-09-04 by BFH-ktt1