logisim-evolution/logisim-evolution #1015: "Uploading" vs "Downloading"

MarcinOrlowski, 2021-09-03:
There's multiple use of word "download"/"downloading" in the project that are incorrectly used. The "downloading" means process of fetching remote data to your local end, while uploading "uploading" is the process of sending local data to remove end, therefore whenever we push anything to the FPGA board  it should be properly called "Uploading".

BFH-ktt1, 2021-09-06:
You are right and not, as it is always from which side you look at it. You "upload" it from logisim, and you "download" it into the FPGA. Unfortunately these two words (as many others) are very often mangled as person A looks at it from point X, while person B interprets it from point Y, where the data goes from X -> Y. Hence in this sense "uploading" may also be wrong and it should be more verbose as "uploading bitfile from logisim to FPGA", which is a lot of words :-) Alternatively: "downloading bitfile into FPGA" also is too verbose :-)

MarcinOrlowski, 2021-09-06:
None the less we should use only one form, not both because that's makes this all quite confusing, as reader do not have have any context to know more. But as in almost any app "downloads" means receiving data on the computer (app) and "upload" is sending out data from the computer/app elsewhere, I suggest to stick to that to for better UX.

R3dst0ne, 2021-09-06:
I agree that we should be consistent with how we call/describe this.
> But as in almost any app "downloads" means receiving data on the computer (app) and "upload" is sending out data from the computer/app elsewhere

This is true for most pieces of software, but in this special case it is not that clear. I regulary read "download on FPGA" in specific programs or literature.

maehne, 2021-09-09:
Maybe, @davidhutchens could comment on this as a native speaker. Maybe, we could also side-step the issue by speaking of "Loading/Programming the bit stream into the FPGA device"

davidhutchens, 2021-09-09:
Since you asked, I'll comment. According to the dictionary on my computer:

upload: computing. verb. transfer (data) from one computer to another, typically to one that is larger or remote from the user or functioning as a server: you can upload your prepared text | software is uploaded and downloaded. Compare with download.

Merriam-Webster dictionary defines download as: an act or instance of transferring something (such as data or files) from a usually large computer to the memory of another device (such as a smaller computer)

That fits with my understanding. Upload typically refers to targeting a larger or remote computer, often a server. So the usage that refers to download to a board is consistent with that definition. You are downloading from the (larger) computer to the (smaller) FPGA. In this case, neither is necessarily remote.

So download is the correct term when targeting an FPGA from a general purpose computer. It doesn't matter that the command is given from the computer end.

Bypassing the issue is a reasonable option.

maehne, 2021-09-09:
Thanks a lot for your input @davidhutchens!

MarcinOrlowski, 2021-09-10:
> So download is the correct term when targeting an FPGA from a general purpose computer. It doesn't matter that the command is given from the computer end.

My understanding + common use case is that "downloading" is action of transferring data to the device that **initiated** the transfer and that simply drives it (and the size of computer should not matter since 1990 :) So if we have PC and FPGA board and if you run LSe on said PC then in case you are sending data from PC to FPGA then, if the whole process is controlled by PC (so FPGA is accepting data but does nothing on its own to get it) then:
1. PC is **uploading** data to FPGA
1. FPGA is **downloading** data from PC
1. Since it's LSe that tells what's going on, therefore as LSe==PC -> "uploading" 

In case the PC (LSe) ints the data transfer by i.e. giving FPGA command with URL to the payload and then do nothing by its own but wait for completion, then definitely we would say "FPGA is downloading". But note that in this case PC is doing nothing really in context of this particular data transfer so you cannot say PC is "uploading". It can only "serve" the data if the URL points to payload stored on that machine.

So in our case as all the messages are produced by LSe (so in other words, LSe talks to us, is our "lyrical subject" :) all messages should be reflecting that and be from LSe "point of view".

R3dst0ne, 2021-09-10:
I don't want to question your English knowledge/skills. But "your understanding" is not important when there are clear definitions.
Additionally, "download to" is commonly used when talking about such things.

Before we start a discussion, I propose that we make a quick poll.
👍🏼: change to upload
👎🏼: keep download
If one side wins clearly, we can skip the discussion and change it or leave it is as it is.

MarcinOrlowski, 2021-09-11:
> I don't want to question your English knowledge/skills.

Why? I would. It's not a hard task. Also -> "Question authority" (Socrates). 

> is not important when there are clear definitions.

Clear definitions by who?

> Additionally, "download to" is commonly used when talking about such things.

Great. As it is commonly used, could you please provide explicit example where fetching data is called "uploading"?

R3dst0ne, 2021-09-11:
>could you please provide a few explicit examples

For FPGAs:
- "Select a bitstream programming file and download it to your hardware device." (Vivado)
- "Downloading Bitstream onto the target board." (XPS)
- "Completed downloading core to device." (iMPACT)
- "Configuration data download to FPGA was not successful." (iMPACT)
- "This custom bitstream is then downloaded to the FPGA." ([MathWorks](https://www.mathworks.com/help/supportpkg/xilinxzynqbasedradio/ug/fpga-targeting-workflow.html))
- "Workflow Advisor Step 4.4 downloads the bitstream onto the device." [MathWorks](https://www.mathworks.com/help/supportpkg/xilinxzynqbasedradio/ug/fpga-targeting-workflow.html))

For MCs:
- "Use the [...] Adapter to download and test program code on your target system." ([KEIL](https://www.keil.com/c51/default.asp))
- "A Terminal window for [...] downloading programs into the memory of the microcontroller using Hex Intel format." ([MC-Tools package](https://www.rathlev-home.de/mc/mctools-e.html))
- "Download code to an Arduino Uno though a specified port." ([WOLFRAM](https://reference.wolfram.com/language/MicrocontrollerKit/tutorial/UploadingPrograms.html)) (This one uses both up- and download.)

closed 2021-09-11 by MarcinOrlowski