logisim-evolution/logisim-evolution #1019: short versions for CLI arguments

R3dst0ne, 2021-09-03:
In #983 short versions for each option get introduced.
@maehne ([here](https://github.com/logisim-evolution/logisim-evolution/pull/983#issuecomment-910488977)) and I ([here](https://github.com/logisim-evolution/logisim-evolution/pull/983#issuecomment-911596004)) are not sure if this is a good idea.
As mentioned in my commend:
> It is better to have no short version than a misleading one.

Currently, there are a view where the letters don't make sense.

I propose to only add short versions for:
`-h`, `--help`
`-v`, `--version`
`-p`, `--plain`
`-e`, `--empty`
`-t`, `--template`
`-g`, `--gates`
`-c`, `--clear-prefs` 
(`-s`, `--substitute` & `-l`, `--load` or `--locale`
are fine for me as well.)

@MarcinOrlowski @BFH-ktt1 @davidhutchens @zdimension @maehne
What are your opinions?

MarcinOrlowski, 2021-09-03:
I am against simply because it is more work than just removing single letter which for now is completely unjustifiable.

BFH-ktt1, 2021-09-03:
Currently we have:
```
Logisim-evolution v3.7.0
https://github.com/logisim-evolution/
develop/1213e23e (2021-09-03T20:32:31+0200)
Java HotSpot(TM) 64-Bit Server VM v16.0.2 (Oracle Corporation)

usage: Logisim-evolution [-b <file>] [-e] [-f] [-g <shape>] [-h] [-j <args>] [-l <file>] [-m
       <geometry>] [-n] [-o <lang>] [-p] [-r] [-s <file>] [-t <format>] [-v] [-w
       <argTestVectorOptionArgName>] [-z <file>]
 -b,--test-circuit <file>                        Open up a circ file and start the test bench within
                                                 it. It returns Success or fail. Arguments:
                                                 <path_to_circ>
 -e,--empty                                      Use empty template.
 -f,--test-fpga                                  Test implementations design using circ_input file
                                                 and optional map_input mapping file (obsolete since
                                                 v3.3.3). The implementation circuit_name is the
                                                 circuit choose to implement (FPGA_Top) and the
                                                 board is the board name located in
                                                 resources/logisim/boards/ without the xml
                                                 extension. The optional tick frequency is in Hz.
                                                 The optional switch HDLONLY will only generate the
                                                 HDL-description and will not perform synthesis,
                                                 P&R, and Download. Arguments: <circ_input>
                                                 [map_input] <circuit_name> <board> [tick frequency]
                                                 [HDLONLY]
 -g,--gates <shape>                              Sets "ANSI" or "IEC" gate style.
 -h,--help                                       Displays this argument summary help page.
 -j,--test-circ-gen <args>                       Open up the circ file and write a new one to later
                                                 check the XML content. Arguments: <circ_input>
                                                 <circ_output>
 -l,--load <file>                                Load image file into RAM (works with -tty only).
 -m,--geometry <geometry>                        Sets geometry for main window: Arguments:
                                                 WIDTHxHEIGHT[+X+Y]
 -n,--no-splash                                  Hides splash screen at startup.
 -o,--locale <lang>                              Sets locale as given as argument.
 -p,--plain                                      Use standard Logisim template.
 -r,--clear-prefs                                Clear application preferences at startup.
 -s,--substitute <file>                          Substitutes library lib1 with lib2. Arguments: lib
                                                 lib2.
 -t,--tty <format>                               Run without graphical interface. Argument: TTY
                                                 format.
 -v,--version                                    Display version number and exit
 -w,--test-vector <argTestVectorOptionArgName>   Test vector. Arguments: <circuit> <test_vector>
 -z,--template <file>                            Use file as template. Argument: file.
```
so the ones which are misleading are -b, -j, -m, -o, -r, -w, and -z.
I'm im favor of the shorts, so maybe we should adapt the "longs" to map the shorts, or adapt some shorts, e.g.:
-b could become -c, which is consistent with -f.

-j -> I'm not sure about what this option does, but I have the feeling that we can drop it

-s -> I'm not sure about what this option does, but I have the feeling that we can drop it

-z --template could become -s --startup-template

-w --test-vector we need to look into details.

I think that this is work in progress and cleanup ready (see also several issues), and hence I would leave for the moment the things as they are and propose to test all the command line options.

I only use extensively -f, so we need to have some feed-back which command line options are used and make sense, and strip all the others.

BFH-ktt1, 2021-09-03:
On another note, -p, -z, and -e are exclusive, so what happens if you specify all three of them, I think that they can be combined into one CLO.

MarcinOrlowski, 2021-09-03:
The logic is ported as it was previously so if there was a check for this case, then it is now too. If not, commons-cli supports group of mutually exclusive options, so it's just matter or setting that up, which should be trivial. If I understand correctly one group is: `-p`, `-z`, `-e`? any other?

BFH-ktt1, 2021-09-03:
@MarcinOrlowski : I understand, but that is not what I wanted to say, my point was that these 3 can combined in one command line option e.g. ```--template``` where you can give as argument "empty", "plain", or "mytemplate.xml", reducing the number of CLO's

BFH-ktt1, 2021-09-03:
If we make it ```--user-template``` than we have directly the short ```-u```

MarcinOrlowski, 2021-09-03:
Can you please create separate ticket for merging these options into new one? Please assign it to me and I will update the parser.

MarcinOrlowski, 2021-09-03:
BTW: I just found pretty neat CLI parser library https://github.com/remkop/picocli/ that offers much more than commons-cli - I will check that in future as we may want to replace commons-cli with it if all the benefits I see it offers can be a reality :)

BFH-ktt1, 2021-09-03:
> CLI access is niche, so I am against simply because it is more work than just removing single letter. Such effort is IMHO pure waste for now.

I just want to give you some insight why for a prof. using logisim the CLI is very important:
1) During the covid period my students were not able to use the real hardware, so we made a setup with hardware boards and a camera, and used the CLI to download their design to the hardware, record for 30 seconds and send them the video, without the CLI this would not be possible.
2) During a practical exam I provided the students (320 in total) with a wrong template file (for a board the did not have), hence I had to download each and every of their circuits to the "correct board", with the gui and clicking I would already be in a "burn-out", using the CLI and some scripting facilitated the task enormously.
3) I do practical exercises where the students solutions are automatically embedded in a test-bench and automatically grades, also here the CLI is very important.

Hence for me the CLI is not a "niche", but a must to be able to automate correction flows.

BFH-ktt1, 2021-09-03:
I now also understand the option -j, this checks if the source-xml-file is not manually modified by somebody after saving to "fool" plagiarism tools, so also this option we could sanatize more in the sense that we make an option that tells us "manually modified", hence returns a true or false instead of a new file

MarcinOrlowski, 2021-09-03:
Ok, so first, I am glad that people use CLI more than I thought. For many ordinary terminal running `ls -l` is hacker wizardry :) And so if CLI is so important, let's take a look at what we have now and lets discuss if we can have something done better. Maybe we should better support  such features or use cases you mentioned? BTW: I once thought that we could add a feature that automatically packs the project, libraries etc and invokes system default mailer

R3dst0ne, 2021-09-03:
> I once thought that we could add a feature that automatically packs the project, libraries etc.

 #13 is about this.

MarcinOrlowski, 2021-09-04:
Also [New save file format #812](https://github.com/logisim-evolution/logisim-evolution/discussions/812) is related

MarcinOrlowski, 2021-09-04:
As per #1024 it's now easy to make short switches optional, so if there is any valid reason why certain options should have long switch only, we can discuss this and apply now with easy.

BFH-ktt1, 2021-09-13:
Can we close this issue? The project export related option is already covered in #13 and #812 .

MarcinOrlowski, 2021-09-13:
Oh sorry, I thought I created this ticket.

R3dst0ne, 2021-09-13:
After the recent changes, the only improvement I can think of, is changing the short option for `--clear-prefs` from `-r` to `-c`.

MarcinOrlowski, 2021-09-13:
I'd not touch `--clear-prefs` at all. Maybe even remove short form completely. Unless I miss anything this is kind of "recovery" action so it won;t be used too often + you do not want to trigger it accidentally, which is far easier with short form. 

I addressed that in #1088

* #1088

R3dst0ne, 2021-09-13:
After `--clear-pref` was adressed in #1088 is there a reason to keep this issue open, or can it be closed?

BFH-ktt1, 2021-09-13:
It can be closed I think

closed 2021-09-13 by BFH-ktt1