logisim-evolution/logisim-evolution #1021: Merge exclusive command line options -p, -z, and -e

BFH-ktt1, 2021-09-03:
As discussed in #1019 the command line options -p, -z, and -e are exclusive and can combined into one.

Proposal:
```--user-template``` with short ```-u``` where the options can be one of the following: "empty", "plain", or "mytemplate.xml"

MarcinOrlowski, 2021-09-07:
What the benefit of merge?

BFH-ktt1, 2021-09-08:
Lesser command line options and having only one for the given functionality.

MarcinOrlowski, 2021-09-09:
These: "empty", "plain", or "mytemplate.xml" are literals? Esp. the last one?

R3dst0ne, 2021-09-09:
`empty` and `plain` are two constants strings that can be detected (if the parameter matches one of them, this option should behave like `-e`/`-p`).
If none of them matches, the argument should be seen as path (like `-z <path>`).

maehne, 2021-09-09:
I am in favour merging the three options. Personally, I like just `--template` best and `-t` would then be a nice short name. In the Logisim-evolution preferences, the option is called "Custom template", so `--user-template` is counter-intuitive. Maybe "circuit template" or "schematic template" would be more descriptive alternatives if a longer name is preferable?

By the way, I noted that the file dialogue to select the template does not restrict the choice to a particular type. Shouldn't this be restricted to `.circ` files?

Of course, we'll also need to adjust the relevant section in the User's Guide.

MarcinOrlowski, 2021-09-09:
I do not like the idea of this merge. These options can be mutually exclusive but that it. First, merge of `-e`/`-p` solves nothing. It makes things worse as you need to remember switch value too. And the idea of argument being filename of option is wrong too. What if I want my file to be named `empty` or `plain`? 

I vote to close this ticket as `wontfix` and create discussion instead to so anyone interested can discuss command line use and then we can design and implement (at least) better thought command line options and once we got that specification designed, we can work on implementation (perhaps using different, more flexible parser).

BFH-ktt1, 2021-09-10:
> What if I want my file to be named `empty` or `plain`?

Then it is correct, as the file would exist, and have an ```.xml``` extension. Hence a simple ```file.exists()``` could distinguish between the ```plain``` as internal template or ```plain.xml``` as external template.

I think the benefits of having lesser exclusive command-line option out weight  the extra intelligence we have to build in.

@maehne : As the template not only includes parts for the circuit, but also the libraries visible, the items in the shortcut-bar, and the mouse shortcuts, the ```--user-template``` is more appropriate. I think that in this case we should rename in logisim ```Custom template``` to ```User template``` to make things consistent.

closed 2021-09-13 by BFH-ktt1