logisim-evolution/logisim-evolution #1022: Drop --no-spash command line option and rename --test-circ-gen

BFH-ktt1, 2021-09-03:
In accordance with the discussion in #1019 the ```--no-splash``` CLO can be dropped (as the splash screen does "not show" anyways on current machines). We can than replace ```--test-circ-gen``` by:
```--new-file-format``` short ```-n``` with two arguments: ```input-file``` and ```output-file``` where the functionality is following:
1) the input file is read and written by the xml-writer to the output file.
2) logisim return an FILES_NOT_IDENTICAL error if the input file is not equal to the output file.
3) logisim return Okay when the files are identical

MarcinOrlowski, 2021-09-03:
> as the splash screen does "not show" anyways on current machines

Actually that's not true - the splash can be seen but for rather short time. I wonder if we can remove it as it serves no real purpose today?

R3dst0ne, 2021-09-03:
> I wonder if we can remove it as it serves no real purpose today?

I would disagree with this. Not everybody has the privilege of owning a good PC.
And this is not limited to private persons. I know schools that use Logisim where 10 -15 year-old PCs are still in use.

MarcinOrlowski, 2021-09-04:
Dropping --no-splash out of scope of this ticket.

BFH-ktt1, 2021-09-04:
> I would disagree with this. Not everybody has the privilege of owning a good PC.
> And this is not limited to private persons. I know schools that use Logisim where 10 -15 year-old PCs are still in use.

@R3dst0ne : I agree with this point, hence I would not drop the splash-screen on startup, I was just wondering who wants to disable this with the command line option (I find this one of the options that make no sense),

MarcinOrlowski, 2021-09-04:
Some people may do not like the filcker they experience on fastest machines as splash implementation is pretty dumb - it opens, something is being set up then it gets closed. Here it is annoying and I'd normally disable it. The other approach (which is very low-pri) would be to open spash with a delay. so if setup is done faster than the threshold, no splash will open (just to be closed). I'd leave --no-splash for now.

MarcinOrlowski, 2021-09-08:
> logisim return Okay when the files are identical

Do we have comparator already? If not, what means "identical" more precisely?

closed 2021-09-13 by BFH-ktt1