logisim-evolution/logisim-evolution #1027: Endless loop in Reporter

BFH-ktt1, 2021-09-04:
I do not know for now how to reproduce with a small example, but I got an endless loop exception as following:
```
Exception in thread "Thread-6" java.lang.StackOverflowError
        at java.base/java.util.Formatter.<init>(Formatter.java:1994)
        at java.base/java.lang.String.format(String.java:3290)
        at com.cburch.logisim.fpga.gui.Reporter.addFatalErrorFmt(Reporter.java:54)
        at com.cburch.logisim.fpga.gui.Reporter.addFatalErrorFmt(Reporter.java:54)
        at com.cburch.logisim.fpga.gui.Reporter.addFatalErrorFmt(Reporter.java:54)
        at com.cburch.logisim.fpga.gui.Reporter.addFatalErrorFmt(Reporter.java:54)
        at com.cburch.logisim.fpga.gui.Reporter.addFatalErrorFmt(Reporter.java:54)
        ....
```

MarcinOrlowski, 2021-09-04:
At first it loks like formatter got a string that maybe had another formatting sequence which caused that? I'd check that first.

> reproduce with a small example

Can I have any reproducible case? I mean steps to repro, project to repro?

BFH-ktt1, 2021-09-04:
I have the feeling that it is a java issue,
This was the code causing the endless loop:
```
      if (subPortIndex < 0) {
        Reporter.Report.addFatalErrorFmt(
            "BUG:  Unable to find pin in sub-circuit\n ==> %s:%d\n",
            this.getClass().getName().replaceAll("\\.", "/"),
            Thread.currentThread().getStackTrace()[2].getLineNumber());
        return false;
      }
```
Basically it should add one line and exit (what it did with java 14), with java16, however, it just kept calling ```Reporter.Report.addFatalErrorFmt(``` (I put a System.out.println() in Report.addFatalErrorFmt()).
As I now fixed the netlist bug I have no reproducible case any more.

BFH-ktt1, 2021-09-04:
I did not see that one :-) Closing, solved

PR #1038 (2021-09-04): Fixed endless loop in AddFataErrorFmt()
 Fixes #1027

* #1027
closed 2021-09-04 by BFH-ktt1