logisim-evolution/logisim-evolution #1042: Poly's moveHandle() always returns null

MarcinOrlowski, 2021-09-04:
In src/main/java/com/cburch/draw/shapes/Poly.java line 319, `moveHandle()` method reads:

```
  @Override
  public Handle moveHandle(HandleGesture gesture) {
    final var hs = getHandles(gesture);
    final var is = new Handle[hs.size()];
    Handle ret = null;
    var i = -1;
    for (val h : hs) {
      i++;
      is[i] = h;
    }
    setHandles(is);
    return ret;
  }
```

my concern is return value as it is always null. And since this is consumed in other parts, I wonder if this is intentional but undocumented, or bug. Anyone knows more?

BFH-ktt1, 2021-09-04:
This is part of Carl Burch's original code that has been never touched. Up to this moment there was no bug report related to it, so I guess that it is okay.

maehne, 2025-09-05:
Should we close this issue due to inactivity? It could be reopened if a real issue due to this behaviour of `moveHandle()` materializes.

maehne, 2026-09-07:
@hewzhew: Thanks for your analysis of this issue! Please go ahead proposing a PR to fix it as outlined. @MarcinOrlowski can then comment on the PR if he likes/finds time.

MarcinOrlowski, 2026-09-07:
@hewzhew i freed the "Assignee" slot, so feel free to fully take this over and handle it as you like. Thanks

closed 2026-09-29 by BFH-ktt1